ros-controls/ros2_controllers #9: As a robotics developer, I want a differential drive controller so that I can control my mobile robot

piraka9011, 2019-12-16:
## Design requirements

Port over the differential drive controller plugin from [`ros-controls/ros_controllers`](https://github.com/ros-controls/ros_controllers/tree/melodic-devel/diff_drive_controller).

BrightenLee, 2020-02-06:
I ported the diff_drive_controller to the ROS2, and I'm using it.
Can I know what will be improved from the previous one?

BrightenLee, 2020-02-06:
Hello, I wonder what will be improved from the previous one(diff_drive_controller)
I ported the diff_drive_controller to the ros2, and I'm using it now.
So I am interested in what changes in ros2.

bmagyar, 2020-02-07:
Great to hear! Care to share?

brightenlee, 2020-02-08:
> Hello, I wonder what will be improved from the previous one(diff_drive_controller)
> I ported the diff_drive_controller to the ros2, and I'm using it now.
> So I am interested in what changes in ros2.

as

brightenlee, 2020-02-08:
> Great to hear! Care to share?
> […](#)

Sure, but it's not perfect yet.

brightenlee, 2020-02-08:
Sure, but it's not perfect yet.

closed 2021-01-02 by bmagyar