ros-controls/ros2_controllers #39: [joint_trajectory_controller] Refactor test cases, smaller scope

bmagyar, 2020-04-29:
Refactor the test cases such that they clearly convey what the code is testing and what the expectation should be. Always avoid if possible, reusing the context of a previous test for simplicity, this is what a fixture class should be for. Example vaguely resembling what's in the test ATM:

```
TEST_F(EmptyFixtureWeCouldHaveFilled, test_sampling)
{
  // various lines setting up the context, should be part of the fixture class

  {
  result = sample(parameters1);
  ASSERT_THAT(result, ...);
  }

  {
  result = sample(parameters2);
  ASSERT_THAT(result, ...);
  }

  {
  result = sample(parameters3);
  ASSERT_THAT(result, ...);
  }

}
```

The problem with this is that the tests can only report "something is wrong with `sample`", we'd still need to dig and look which inner case and why. To aide in this, the pattern above is sometimes augmented with comments for each context block. I've done a lot of this and found it quite bad.

Instead, consider something like this:
```
TEST_F(TestSampling, works_with_empty)
{
  result = sample(parameters1);
  ASSERT_THAT(result, ...);
}

TEST_F(TestSampling, fails_with_time_from_start_undefined)
{
  result = sample(parameters2);
  ASSERT_THAT(result, ...);
}

TEST_F(TestSampling, success_when_all_params_defined)
{
  result = sample(parameters3);
  ASSERT_THAT(result, ...);
}
```
The result when running the tests will show a lot more cases and be a lot more specific with failures. Also, note that in this case I used a specific class fixture for `TestSampling` only, it is good practice to use the fixture name to convey the general area we are testing so we don't have to duplicate it in the name of the case, e.g `TEST_F(TestJTC, sample_success_when_all_params_defined)`

bmagyar, 2020-04-29:
FYI @Karsten1987 @ddengster

jordan-palacios, 2020-06-10:
I'll be working on this.

PR #64 (2020-06-18): Refactor test cases
Refactoring joint_trajectory_controller tests to fix #39.

I'm doing some cleanup since I'm at it.
closed 2020-06-18 by bmagyar