ros-controls/ros2_controllers #40: [joint_trajectory_controller] Improve clamping code

bmagyar, 2020-04-29:
`trajectory.cpp`:
```
percent = percent > 1.0 ? 1.0 : percent;
percent = percent < 0.0 ? 0.0 : percent;
```
could be written as
```
percent = std::clamp(percent, 0.0, 1.0); // C++17
```
or
```
percent = boost::algorithm::clamp(percent, 0.0, 1.0); // boost
```.

destogl, 2020-05-05:
IMHO: better to use `std` instead of `boost`

bmagyar, 2020-05-07:
Absolutely! I listed boost simply out of my own ignorance on what is currently available on non-Ubuntu platforms

v-lopez, 2020-06-11:
All ros2_control packages I've seen are targeting C++14, so I'm not sure we can use `std::clamp`.

But they are not using Boost either, so I'm not sure it's worth adding it just for that.

We can do: `percent = std::min(1.0, std::max(0.0, percent));` but I would not say it is better in any way.

bmagyar, 2020-06-12:
Should we create a header we can copy around that hosts some of these simple helpers? We are already hosting some junk headers (visibility, looking at yous!) so an actual utility would be cool to have there.
A note could be added to this file for each function when they could be replaced (moving to C++17, etc)

v-lopez, 2020-06-17:
Another one for the refactor on diff_drive: https://github.com/ros-controls/ros2_controllers/commit/5bace65938c1bb0541f13810634dae08ca9892b2#diff-2ca4938a2f42b5166cf293e314b92fadR44

v-lopez, 2020-06-17:
Related to diff_drive, it is using C++17. So in that one we could actually use std::clamp.

v-lopez, 2020-07-07:
Diff drive is no longer using C++14.

So what do we do? A header somewhere common like `controller_interface`?
I don't like copying any header around.

Karsten1987, 2020-07-07:
can we contribute things like clamp to [rcpputils](https://github.com/ros2/rcpputils/tree/master/include/rcpputils)? We've already have things like `split` in there and a C++14 backport of `std::filesystem`. I believe `std::clamp` could nicely live there as well.

bmagyar, 2020-07-08:
Great idea, let's do that!

v-lopez, 2020-07-08:
Alright, I'll look into it.

closed 2020-07-18 by bmagyar