ros-controls/ros2_controllers #41: [joint_trajectory_controller] Review sharedptr pass by const reference practice

bmagyar, 2020-04-29:
@Karsten1987 @ddengster take it away ;) 

Either way I think we should be consistent in style and currently there's definitely at least one outstanding item. Let's address it.

v-lopez, 2020-06-18:
Regarding this subject I always suggest following rules [R34, R35 and R36](https://github.com/isocpp/CppCoreGuidelines/blob/master/CppCoreGuidelines.md#r34-take-a-shared_ptrwidget-parameter-to-express-that-a-function-is-part-owner) 

Which basically means, if you don't need to interact with the ownership of the object, don't pass a smart pointer, pass a reference.

Karsten1987, 2020-06-18:
Can you bring examples where we pass a shared pointer by const ref?
Without precise hints on where you see an inconsistency here, I don't think there's much to "take away" from this issue.

v-lopez, 2020-06-19:
On a closer look I haven't found a  lot. 

It's true that if you compare with ROS1 code there's signifcantly more shared_ptrs, but I've found a reasoning for them, except a few on tests and this one: 
https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/include/joint_trajectory_controller/tolerances.hpp#L98

closed 2020-07-29 by bmagyar