ros-controls/ros2_controllers #58: Revise for-loop style

bmagyar, 2020-06-05:
`for (auto i = 0ul; i < goal->trajectory.joint_names.size(); ++i)`

could be written as

`for (std::size_t i = 0; i < goal->trajectory.joint_names.size(); ++i)`
or 
`for (size_t i = 0; i < goal->trajectory.joint_names.size(); ++i)`

which avoids using a *typed literal and auto* in favour of the classic *type and type-agnostic literal* which improves readability.

Any takes on this? @v-lopez perhaps?

v-lopez, 2020-06-05:
I'd go with the `size_t i = 0` version. 

I can go over the code and look for these. I'm currently working on https://github.com/ros-controls/ros2_controllers/issues/38 and I can take this afterwards.

vasantteja, 2020-06-05:
Can I get this is issue assigned to me?

Timple, 2021-10-12:
If the index is only used to get the element, this:

```c++
for (std::size_t i = 0; i < goal->trajectory.joint_names.size(); ++i) {
  std::string joint_name = goal->trajectory.joint_names[i];
  do_stuff(joint_name);
}
```
Can also be written like the readable:

```c++
for (auto joint_name : goal->trajectory.joint_names) {
  do_stuff(joint_name);
}
```

closed 2021-10-15 by bmagyar