ros-controls/ros2_controllers #59: Refactor test "mini simulator"

bmagyar, 2020-06-05:
Our integration tests currently have a recurring pattern where the robot hardware read/update/write steps are repeated a couple of times in a loop until a condition is met.

An occurence of this pattern is such:
```
  auto start_time = rclcpp::Clock().now();
  rclcpp::Duration wait = rclcpp::Duration::from_seconds(0.5);
  auto end_time = start_time + wait;
  while (rclcpp::Clock().now() < end_time) {
    test_robot->read();
    traj_controller->update();
    test_robot->write();
  }
```
To improve readability and reduce duplicated code in the future, this could be factored into a test_common file with a name along the lines of `run_mini_simulation(test_robot, time_to_run_for)`.

bailaC, 2021-10-20:
I see the code in 3 files, but they are somewhat modified.

```
src/ros2_controllers/joint_trajectory_controller/test/test_trajectory_actions.cpp:
  94        auto end_time = start_time + wait;
  95:       while (rclcpp::Clock().now() < end_time)
  96        {
```

```
src/ros2_controllers/joint_trajectory_controller/test/test_trajectory_controller_utils.hpp:
  303      const auto end_time = start_time + wait_time;
  304:     while (rclcpp::Clock().now() < end_time)
  305      {
```

```
src/ros2_controllers/joint_trajectory_controller/test/test_trajectory_controller.cpp:
  404    const auto end_time = start_time + wait;
  405:   while (rclcpp::Clock().now() < end_time)
  406    {
```

Ishan1923, 2026-01-22:
Hi @christophfroehlich, I would like to work on this issue. I will start by identifying where this loop pattern currently exists in the codebase. Please assign it to me.