ros-controls/ros2_controllers #80: message data type for cmd_vel in diff_drive_controller

rjshim, 2020-07-16:
Hi,

From those two files ([diff_drive_controller.hpp](https://github.com/ros-controls/ros2_controllers/blob/master/diff_drive_controller/include/diff_drive_controller/diff_drive_controller.hpp#L51), [diff_drive_controller.cpp](https://github.com/ros-controls/ros2_controllers/blob/master/diff_drive_controller/src/diff_drive_controller.cpp#L395)), you can see that diff_drive_controller is using TwistStamped instead of Twist that is used by many wheel-mobile robot-related applications such as the navigation stack and many teleop examples. 

I wonder if time information for twist message is necessary because potential users will all need to make a bridge node subscribing to cmd_vel and regenerating a topic message with a new type TwistStamped including time information.

Ryan

bmagyar, 2020-07-16:
Switching over is not a problem, however I'd find it pretty strange for the nav stack to drop the header from their command messages. Do you have a reference for this?

rjshim, 2020-07-17:
Thanks for your reply.

[This](https://github.com/ros-planning/navigation2/blob/master/nav2_controller/src/nav2_controller.cpp#L109) is for the navigation stack and [this](https://github.com/ros-planning/navigation2/blob/master/nav2_controller/src/nav2_controller.cpp#L352) is for the basic teleop keyboard pacakge by OSRF. 

Though I also thought it is a good idea to filter old data like [this](https://github.com/ros-controls/ros2_controllers/blob/master/diff_drive_controller/src/diff_drive_controller.cpp#L203). It doesn't necessarily have to be Twist instead of TwistStamped, but it will be great if at least there is an agreement on which message type will be used for the cmd_vel topic.

bmagyar, 2020-07-17:
I see. I think that ignoring the timestamp of a command in a robotic system and expecting it to work reliably is pretty naive. This may work ok within toy examples with the nav stack provided that these messages are all taken care of internally, never leading those nodes. The references you showed already show use of TwistStamped all around it. In any realistic system you'd want those header in my opinion.

rjshim, 2020-07-20:
@bmagyar 

Thank you for the comment. :)
I agree. It will be better and more accurate to design the controller as you said. However, compatibility wise, considering the fact that most people use ROS because it is easy to build up their own robot system by simply downloading and running individual packages (ROS [plumbing](https://www.google.com/search?q=ROS+plumbing&source=lnms&tbm=isch&sa=X&ved=2ahUKEwiftp7MhdvqAhXFw4sBHXDmBKEQ_AUoAXoECBgQAw&biw=1321&bih=932#imgrc=zT8mOLLikSDJxM)), perhaps it needs further consideration. As you said, the nav2 stack uses TwistStamped internally, but the final input/output type of topic messages for many current available applications including the nav2 stack is Twist. This is [diff_drive_controller](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_plugins/src/gazebo_ros_diff_drive.cpp#L139) for Gazebo in ROS2.

I think adding a YAML file for parameter setting and making it easy to switch between using Twist and TwistStamped is a possible option without making too much modification. 

Ryan

bmagyar, 2020-08-03:
Let's see the flip side of this tortilla!

closed 2020-08-03 by bmagyar

SteveMacenski, 2020-08-10:
> however I'd find it pretty strange for the nav stack to drop the header from their command messages

We can't drop what we never had :rofl: 

I support changing ros2_control, nav2, gazebo plugins, turtlebot3 pkgs, robot_localization, etc, etc, etc to `Stamped` but we need to change that API between distributions, not in the middle of one. We can target Galactic, so we should start now with a full list of things we need to update. https://github.com/ros-planning/navigation2/issues/1594 has a couple of preliminary lists.

rjshim, 2020-08-11:
@SteveMacenski Or maybe keeping the current "cmd_vel" topic and adding a new topic called "cmd_vel_stamped" as you said [here](https://github.com/ros-planning/navigation2/issues/1594). 

@bmagyar As Foxy is currently the longest LTS version in ROS2, I hope ros2_controller to also support cmd_vel without time stamps. I can add a parameter "use_time_stamp", make minimum changes to the current source code, and send a pull request.

SteveMacenski, 2020-08-11:
> I hope ros2_controller to also support cmd_vel without time stamps

It better, there's literally nothing else in the ecosystem this will work with for at least 3 years then :stuck_out_tongue:

rjshim, 2020-08-17:
@bmagyar 

Just a kind reminder. :)

Ryan