ros-controls/ros2_controllers #82: Implement Twist controllers (rather than TwistStamped) for mobile robot uses

SteveMacenski, 2020-08-14:
Hi,

I see this remap from Twist -> TwistStamped in the header files for using the Stamped version of messages. While I support using Stamped messages long term, right now no other item in the ecosystem will be compatible with this (gazebo, nav2, etc) so we should be supporting Twist right now in Foxy and older. For Galactic and newer, we can target TwistStamped but it would require a coordinated effort to change that API for a newer distribution.

piraka9011, 2020-08-26:
I'll start looking into the port 👀

rjshim, 2020-08-27:
@piraka9011 

Thank you very much. Please refer to the comment that I gave on this issue ([link](https://github.com/ros-controls/ros2_controllers/issues/80#issuecomment-671664898)). It will minimize the workload as well as support both message types (Twist and TwistStamped).

Ryan