ros-controls/ros2_controllers #110: Can we rename joint_state_controller

destogl, 2020-11-16:
I propose to remove "_controller" suffix from non-controllers.
This will first and foremost influence the 'joint_state_controller', and than also the others, e.g. 'force_torque_sensor_controller'

The simplest proposal is to use word 'publisher' instead of 'controller':

'joint_state_publisher'
'force_torque_publisher'

Or ideas:
* 'observer' (also used ins control theory)
* 'reporter'

What do you think?
Do you have any other ideas?

bmagyar, 2020-11-16:
Absolutely! This was something that we were planning to do. Originally I even thought about a new "PublisherInterface" that'd only allow grabbing state interfaces, never command ones.

destogl, 2020-11-16:
> Originally I even thought about a new "PublisherInterface" that'd only allow grabbing state interfaces, never command ones.

This sounds interesting! Good idea!
We should probably rename it just after #109 is merged, and dann add an interface in the follow-up.

gavanderhoorn, 2020-11-16:
@destogl wrote:
> The simplest proposal is to use word 'publisher' instead of 'controller':
> 
> 'joint_state_publisher'

renaming :+1:, but `joint_state_publisher` is an exiting node, and seeing as we already have quite some confused people when it comes to `joint_state_publisher` and `joint_state_controller` right now (see [#q303358](https://answers.ros.org/question/303358) on ROS Answers fi), it might be good *not* to use `joint_state_publisher` for something in `ros_control`.

olivier-stasse, 2020-11-18:
I think a lot of group have think which are state estimator and which are usually in the controller graph.
They are not simply sensor, but from the control viewpoint they are observer.

destogl, 2021-03-24:
In the discussion in the [WG Meeting on 24th March 2021](https://docs.google.com/document/d/1818AoYucI2z82awL_-8sAA5pMCV_g_wXCJiM6SQmhSQ/edit#heading=h.58xl8zrocn7x) it is decided to go with **broadcaster**

bmagyar, 2021-03-25:
Closing this one, opened a task for this

closed 2021-03-25 by bmagyar