ros-controls/ros2_controllers #116: Take care for documentation

destogl, 2020-11-30:
@bmagyar This assumes that command_interfaces_ comes ordered as you requested it, but I did not find any documentation explicitly mentioning this, that's why I went with the `get_ordered_interfaces` thing.

we could just explicitely mention that the itnerfaces are provided in the order they were requested, that would simplify things.

_Originally posted by @v-lopez in https://github.com/ros-controls/ros2_controllers/pull/99#discussion_r532390821_

saikishor, 2025-07-15:
> @saikishor I tried to deprecate `get_ordered_interfaces` and remove its occurrences. But it is used by 
> assign_loaned_state_interfaces and assign_loaned_command_interfaces of the [semantic components](https://github.com/search?q=repo%3Aros-controls%2Fros2_control%20assign_loaned_state_interfaces&type=code) and its [implementations](https://github.com/search?q=repo%3Aros-controls%2Fros2_controllers%20assign_loaned_state_interfaces&type=code). Can we get rid of this method as well? There exists tests if the jumbled order is fixed, but as you said we can assume that the order is the same as state_interface_configuration and command_interface_configuration?

Maybe we should leave it around, but just remove the implementations?

saikishor, 2025-07-15:
> that's what I am working on. as we discussed, there is a need of get_ordered_interfaces for the semantic components, but not for standard controllers as we do now (see the merged PR in the demos).

Yess! Perfect

closed 2025-08-13 by bmagyar