ros-controls/ros2_controllers #119: Let's restore the CI (again)

bmagyar, 2020-12-01:
@destogl @v-lopez I managed to update things to the right version so that linters don't fail anymore but we have the boost::shared_ptr pluginlib issue again even though I have the flags set... Have they changed in foxy?

v-lopez, 2020-12-02:
I'm not aware of the issue. What is it about?

bmagyar, 2020-12-02:
Note the non-CI changes here: https://github.com/ros-controls/ros2_controllers/pull/104

The error reported by the CI is 
```
  --- stderr: joint_state_controller
  In file included from /root/upstream_ws/install/controller_manager/include/controller_manager/controller_manager.hpp:37,
                   from /root/target_ws/src/ros2_controllers/joint_state_controller/test/test_load_joint_state_controller.cpp:18:
  /opt/ros/foxy/include/pluginlib/class_loader.hpp:55:10: fatal error: boost/shared_ptr.hpp: No such file or directory
     55 | #include <boost/shared_ptr.hpp>
        |          ^~~~~~~~~~~~~~~~~~~~~~
  compilation terminated.
  make[2]: *** [CMakeFiles/test_load_joint_state_controller.dir/build.make:63: CMakeFiles/test_load_joint_state_controller.dir/test/test_load_joint_state_controller.cpp.o] Error 1
  make[1]: *** [CMakeFiles/Makefile2:155: CMakeFiles/test_load_joint_state_controller.dir/all] Error 2
  make: *** [Makefile:141: all] Error 2
  ---
```

I'm pretty sure the fix is something trivial but it's a needle in a haystack for me ATM :D

gavanderhoorn, 2020-12-02:
Is `boost` listed as a build dependency?

bmagyar, 2020-12-02:
We don't have boost and have explicit flags for pluginlib to not try using boost.

gavanderhoorn, 2020-12-02:
Yeah, I figured. It was a stupid comment to post.

destogl, 2020-12-02:
You have to have the variable set also in all packages which use pluginlib (directly or indirectly).

This means all controller will probably have to have this...

bmagyar, 2020-12-24:
Resolved by #122

closed 2020-12-24 by bmagyar