ros-controls/ros2_controllers #129: Redeclaration of joint trajectory controller parameters

livanov93, 2020-12-17:
While testing joint trajectory controller I have encountered an error regarding redeclaration of parameters. 
I think it is done twice, in[ controller interface](https://github.com/ros-controls/ros2_control/blob/ec8b0066685ce24ce28eaf439b0b353207bb6bb1/controller_interface/src/controller_interface.cpp#L27) (base class) and in [joint trajectory controller itself](https://github.com/ros-controls/ros2_controllers/blob/65874894b4f6db1b6075f642f2c5dab8d5c68882/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L66).

The parameters were loaded via this file:
`controller_manager:
  ros__parameters:
    update_rate: 100  # Hz

    forward_command_controller_position:
      type: forward_command_controller/ForwardCommandController

    joint_state_controller:
      type: joint_state_controller/JointStateController

    ur_joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController`

ur_joint_trajectory_controller:
  ros__parameters:
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint
    state_publish_rate: 100.0
    action_monitor_rate: 20.0
    allow_partial_joints_goal: false
    constraints:
      stopped_velocity_tolerance: 0.01
      goal_time: 0.0

bmagyar, 2020-12-17:
Can you give me steps to reproduce please?

livanov93, 2020-12-17:
```
cd ~/catkin_ws/src

git clone https://github.com/PickNikRobotics/Universal_Robots_ROS2_Driver.git --branch livanov/built_version

# you should probably remove moveit repositories from the src/Universal_Robots_ROS2_Driver/.repos.yaml to speed up building

# Clone source-based dependencies into src/ directory
vcs import --skip-existing --input src/Universal_Robots_ROS2_Driver/.repos.yaml src

# ignore ur controllers - fts controller needs update
touch ~/catkin_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/COLCON_IGNORE

# Install package-based dependencies
rosdep install -y --rosdistro $ROS_DISTRO --ignore-src --from-paths src

# Build sources
source /opt/ros/$ROS_DISTRO/setup.bash
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release

source ~/catkin_ws/install/setup.bash

roslaunch ur_robot_driver ur5_system_position_only.launch.py

```

bmagyar, 2020-12-19:
I am working through the release ATM but will get back to you on this once I tried. 
I realized we may also have issues when unloading and re-loading the same controller with the same error that I think we can solve with a combination of `undeclare_parameter()` + `get_parameters(names)` in the right spot.

[image]

bmagyar, 2020-12-24:
@livanov93 could you please check this & close the issue if it has been resolved?

livanov93, 2020-12-30:
Solved with [276](https://github.com/ros-controls/ros2_control/pull/276)

closed 2020-12-30 by livanov93