ros-controls/ros2_controllers #135: [joint_trajectory_controller] Velocity control is unfinished?

Jarijaas, 2020-12-31:
Hey,

So I have been testing this controller with my robot. When I set `time_from_start: { sec: 5, nanosec: 0 } }`, traversing the trajectory should take 5 seconds, which is true, but the velocity is incorrect. The controller correctly set's the positions, but because the controller does not set write new velocity values to the  command interface, the traversed trajectory is unsmooth; the joint moves too fast, so the joint movement stops before the new intermediate position is written.

Shouldn't the velocity be set using the command interface or did I misunderstood something?

I believe, that the velocity value should be set around [here](https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L169-L187)

destogl, 2021-01-01:
Hi @Jarijaas,

this depends on the used hardware (robot). probably your robot is trying to drive maximal velocity (shortest time) between two positions. The current implementation of `joint_trajectory_controller` enables only position interface toward hardware ([see line#83](https://github.com/ros-controls/ros2_controllers/blob/b0e3d198667d885f8a0fbd46c02a247ffdeb85fb/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L83)). I believe this is the same functionality as in ROS1.

How do you expect that the controller behaves:
* set position and velocity for hardware, or, 
* set only velocity in the hardware.

Also, which hardware, i.e. interface implementation are you using?

Jarijaas, 2021-01-03:
Hey @destogl,

My hardware interface is based on the [demo](https://github.com/ros-controls/ros2_control_demos/tree/master/ros2_control_demo_hardware), I added velocity interface to it, and I use it to control dynamixel servos.

I would expect that the `joint_trajectory_controller` should support setting the desired velocity value for the joints using the hardware interface.  I'm probably going to fork this controller and add velocity hardware command interface support to it, as this would suit my needs.

gavanderhoorn, 2021-01-03:
@destogl wrote:
> The current implementation of `joint_trajectory_controller` enables only position interface toward hardware ([see [...]

actually, there are 5 variants of the JTC, supporting position, velocity, effort, position & velocity and position, velocity & acceleration interfaces. See [joint_trajectory_controller.cpp](https://github.com/ros-controls/ros_controllers/blob/1a085c37d212d1b52f44b8b1ebb6167cef90cc65/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L36-L89).

I believe @Jarijaas' question is justified: it seems the current implementation in `ros2_control` is incomplete, at least when compared to the ROS 1 version.

destogl, 2021-01-05:
@gavanderhoorn thanks for pointing this. We should than extend this.

@Jarijaas would you try to port this from ROS1?

We can look at this, but it will take some time to get to it...

Jarijaas, 2021-01-08:
@destogl I don't have time to do it in the near future either.

christophfroehlich, 2023-09-09:
@destogl Is there something missing with the current state of JTC or can we close this in favor of #171?

bmagyar, 2023-09-11:
Definitely covering this already

closed 2023-09-11 by bmagyar