ros-controls/ros2_controllers #151: Use methods in all controller in the same order (Let's make controller development a joy)

destogl, 2021-02-25:
Hi,

Working on an extension of JTC (#140), it's very disturbing (or at least disconcerting) not having unified members/methods order between the different controllers and the controller_interface. The issue I have is jumping between different states of the controller and having to check all the time at which lifecycle state controller is, i.e., which members are filled and which not...

I propose to restructure controller_interface methods to the execution order and keep that throughout this repository. 

P.S. I know this is a very unproductive request. Still, IMHO it's important to increase understanding and finally acceptance of the library...

Any thoughts?

destogl, 2021-03-03:
I propose the following order:

- constructor
- init
- configure()
- command_interface_configuration()
- state_interface_configuration()
- activate
- update
- deactivate
- cleanup
- shutdown

christophfroehlich, 2023-08-16:
I've just read this issue for the first time, and agree with you ;) 
on_error() is new since 2021, but this can go to the end