ros-controls/ros2_controllers #153: Not able to load forward_command_controller

bailaC, 2021-03-07:
Hey All,

While running `ros2 control load_configure_controller forward_command_controller`, getting below error,

> `[ros2_control_node-1] [ERROR] [1615107992.901048982] [controller_manager]: 'type' param not defined for 'forward_command_controller'.`

Any help here? I am raising this as issue because it seems some bug in the code. Honestly I couldn't find any!!

shonigmann, 2021-06-28:
Perhaps a bit late for you @bailaC, but for anyone new I have seen this issue pop up when there is an inconsistency in the naming between the .yaml config file used to start the controller_manager (e.g. [here](https://github.com/ros-controls/ros2_control_demos/blob/bf32f38d4d3dca02bde7d46ac15af51fe4870c66/ros2_control_demo_bringup/config/rrbot_multi_interface_forward_controllers.yaml#L5-L21)) and what is used on the commandline. 

So make sure your .yaml file includes:
```
controller_manager:
  ros__parameters:
    ...
    forward_command_controller:
      type: forward_command_controller/ForwardCommandController
```

and that the `ForwardCommandController` plugin exists as an available controller

bailaC, 2021-06-29:
Thanks a lot @shonigmann for the explanation. This problem was fixed long back. And you said correctly that we should be careful about the name we use in .yaml config and the same must be used while starting the controller.

e.g. if anyone has something like below in .yaml file,
```
controller_manager:
  ros__parameters:
    ...
    my_fancy_controller:
      type: forward_command_controller/ForwardCommandController
```

the controller must be started with the name `my_fancy_controller`, not `forward_command_controller`. And of course what you said 

> that the `ForwardCommandController `plugin exists as an available controller

Closing this issue.

closed 2021-06-29 by bailaC