ros-controls/ros2_controllers #157: Create alternative to std_msgs/Float64MultiArray in ForwardCommandController & family

bmagyar, 2021-03-16:
# Problem:
std_msgs/Float64MultiArray is deprecated.

```
# This was originally provided as an example message.
# It is deprecated as of Foxy
# It is recommended to create your own semantically meaningful message.
# However if you would like to continue using this please use the equivalent in example_msgs.

# Please look at the MultiArrayLayout message definition for
# documentation on all multiarrays.

MultiArrayLayout  layout        # specification of data layout
float64[]         data          # array of data
```

https://github.com/ros2/common_interfaces/blob/foxy/std_msgs/msg/Float64MultiArray.msg

# Proposal:
Create (an) alternative in `control_msgs`. This could be a set of messages with a single joint interface type, one supporting multiples, or one message that has a field for the joint interface type and one that is fully dynamic --- (dynamic joint states??).

gavanderhoorn, 2021-03-16:
> std_msgs/Float64MultiArray is deprecated.

yeah, this really should have been given some more attention.

See [SOSS: a whole new approach to your ROS 1-ROS 2 bridge!](https://discourse.ros.org/t/soss-a-whole-new-approach-to-your-ros-1-ros-2-bridge/17040/8) where this was casually dropped in an off-topic discussion. "It's in the release notes, we can't do anything more" (which is true, it's mentioned in [the notes](https://docs.ros.org/en/foxy/Releases/Release-Foxy-Fitzroy.html#std-msgs), but I doubt many people have seen those).

bmagyar, 2021-03-22:
Agreed. I don't think I've ever read the release notes of a ros distro 😓

destogl, 2021-03-22:
Yeah, I also had no idea, and although I looked at this message a few times, I didn't read its header. Actually, a ROS newcomer I was helping pointed me into this...

This should be a massive topic on ROS discourse...

gavanderhoorn, 2021-03-22:
Feel free to bring it up again.

I've directly asked for an announcement. The response was:

> It's in the release notes, we can't do anything more

ijnek, 2021-09-28:
Mentioned in [Discourse Post](https://discourse.ros.org/t/msg-type-for-publishing-jointcommands/22451)

RhysMcK, 2023-05-15:
Any update on this?

bmagyar, 2023-05-15:
Well it's deprecated but still around... If you want to introduce something for this yourself you are welcome to submit a PR to `control_msgs`

BrittanyKCohen96, 2025-07-30:
Thank you! This sounds really good, that was exactly the issue that we had when we were using it. I'm in final exams but I'll get started on the 19th of August.