ros-controls/ros2_controllers #158: Create joint_state_broadcaster to replace joint_state_controller

bmagyar, 2021-03-25:
We've recently decided to suffix all not-really-controllers (joint_state_controller, force_torque_controller, etc from ROS1) with *broadcaster* to follow the nomenclature of tf and to avoid name clashes with publishers (ROS) and observers (control theory). 

Task is considered complete when
* there's a new package `joint_state_broadcaster`, with all references to `joint_state_controller` gone
* the `joint_state_controller` package still exists but includes a very minimal amount of code to make the old name still available
* add a deprecation note to plugin.xml and other visible places (if any) to `joint_state_controller`

Supersedes #110

matthew-reynolds, 2021-03-25:
I don't have permissions in this repo to self-assign, but I will take this on.

matthew-reynolds, 2021-05-18:
I think this can be closed

closed 2021-07-30 by Karsten1987