ros-controls/ros2_controllers #163: How to setup diff_drive_controller?

Schulze18, 2021-03-29:
Hello everyone, congratulations on the amazing work with ROS1 and ROS2 control. 

We are migrating our robot to ROS2 . For a first test, I am trying to use diff_drive_controller from ROS2 Control at Gazebo. I setup the ros2_control tags and the gazebo_ros2_control plugin the same way presented in the [ros2_control_demos](https://github.com/ros-controls/ros2_control_demos). It worked fine for position_controllers and velocity_controllers, but it did not work for the diff_drive_controller. I am having the following error message:

`[gzserver-1] [INFO] [1617050376.482951728] [controller_manager]: Loading controller 'diff_drive_controller' `
`[gzserver-1] terminate called without an active exception `
`[gzserver-1] Aborted (core dumped) `
`[ERROR] [gzserver-1]: process has died [pid 80076, exit code 134, cmd 'gzserver /PATH_TO_WORLD_FILE                                                                   -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so `

I tried to find an example of it, but I did not find it. Is there any example package using this controller? 

I am attaching the parts related to the wheels transmissions, ros2_control, and controller parameters. Thank you.

[robot_controller.yaml.txt](https://github.com/ros-controls/ros2_controllers/files/6224562/robot_controller.yaml.txt)
[ros2_control.urdf.xacro.txt](https://github.com/ros-controls/ros2_controllers/files/6224563/ros2_control.urdf.xacro.txt)
[wheels.transmission.xacro.txt](https://github.com/ros-controls/ros2_controllers/files/6224564/wheels.transmission.xacro.txt)

bmagyar, 2021-03-30:
I did not spot anything wrong on the first look at these files.
Have you tried this demo implementation? https://github.com/ros-controls/ros2_control_demos/pull/70/files

Schulze18, 2021-03-30:
Thank you, I have not found this PR. I launch it, and it seems to be running, but there is no topic publishing/subscribing to any command. I was able to adapt it to my robot, but the same behavior occurred.

In the codes that I sent, I think the problem is the interaction with the hardware plugin gazebo_ros2_control/GazeboSystem. Is this one the right plugin to use with diff_drive_controller? Or maybe the problem is the ros2_control_demo_hardware/DiffBotSystemHardware?

From what I remember from ROS1, the gazebo_ros_control/DefaultRobotHWSim would expose all the joints with transmission to any ros controller.

In ROS2, I understood that all the hardware interfaces are selected at ros2_control tags in URDF. As I consequence, it seems that the hardware interfaces need to "talk" to the gazebo_ros2_control. Is this the right workflow?

christophfroehlich, 2024-04-04:
I'll close this issue because there are several tutorials or example out there nowadays, e.g. first-hand
- https://control.ros.org/master/doc/gazebo_ros2_control/doc/index.html#gazebo-ros2-control-demos
- https://control.ros.org/master/doc/ros2_control_demos/example_2/doc/userdoc.html

closed 2024-04-04 by christophfroehlich