ros-controls/ros2_controllers #188: Release for Galactic and Rolling

henningkayser, 2021-05-25:
What's the status on releasing `ros2_controllers` for Galactic or Rolling?
We are planning to [release MoveIt](https://github.com/ros-planning/moveit2/issues/476) for both distros end of June and it would be great if this dependency would be available by then.

Timple, 2021-06-08:
Did you test this from source already with MoveIt?

I ran into some issues and I'm wondering if you encountered similar ones: https://github.com/ros-controls/ros2_controllers/pull/199

vatanaksoytezer, 2021-06-14:
@Timple sorry, just saw your message. We've been using ros2_controllers in Rolling and Galactic. We've also set up a new CI for moveit2 build that tests Galactic and Rolling. See https://github.com/ros-planning/moveit2/pull/494. ros-controls people helped us merge a couple of PRs to enable Rolling and Galactic source builds already. If you can share what issues you are facing, I would be glad to help.

Timple, 2021-06-15:
The issues I'm facing are with the `diff_drive_controller`. Probably not in use for moveit :slightly_smiling_face: 
But I already solved them in #199

joe28965, 2022-04-12:
Is it me or has ros2_controllers since been released on both Galactic and Rolling and should this issue be closed?

bmagyar, 2022-04-12:
Oh yes of course, long since it's been released to both. Closing

closed 2022-04-12 by bmagyar