ros-controls/ros2_controllers #194: Request: ackermann_steering_controller

Timple, 2021-06-04:
Feature request: Ackermann steering controller

It seems it was available in [ros1](https://github.com/ros-controls/ros_controllers/tree/noetic-devel/ackermann_steering_controller). However the implementation seemed [off](https://github.com/ros-controls/ros_controllers/issues/412). 

Also support for front-wheel drive seems to be [missing](https://github.com/ros-controls/ros_controllers/blob/a075bc36ebf5b81a7dafe911907cc1b781dd0185/ackermann_steering_controller/src/ackermann_steering_controller.cpp#L334). The front wheels have a different radius they drive about, so the velocity is slightly different. But in case of e.g. a forklift this becomes significant.

Timple, 2021-06-15:
I'm rather new to `ros2_control` and this requires several interfaces on velocity and position level.

If somebody is willing to give a head-start by providing the cpp/hpp dummy classes, I'm happy to finish the work with all logic/algorithms involved!

bmagyar, 2021-06-15:
I think the work here would be porting the existing controller to ros2. The tests are where a lot of complexity lies but good news is that they are basically copies of the diff drive tests. Bad news is that we never got around to porting "the good stuff" when it comes to tests.

destogl, 2021-08-09:
> If somebody is willing to give a head-start by providing the cpp/hpp dummy classes, I'm happy to finish the work with all logic/algorithms involved!

Are you still interested into this? I could find some time to set up initial files and tests.

Timple, 2021-08-09:
Yup, this would still be on the 'todo' list.

Altough it also seems that it would just be a subset of this: https://discourse.ros.org/t/independent-swerve-steering-controller-for-ros-control/21534/1

Eitherway having the dummy classes and tests would really help!

tomlogan501, 2021-11-08:
+1 for swerve controller and four wheel controller

samuk, 2022-02-14:
> > If somebody is willing to give a head-start by providing the cpp/hpp dummy classes, I'm happy to finish the work with all logic/algorithms involved!
> 
> @Timple Are you still interested into this? I could find some time to set up initial files and tests.

@destogl do you think you may still find time for this?

christophfroehlich, 2023-09-29:
I think the requested controller was added with #484. If something is missing, please open a more specific issue.

closed 2023-09-29 by christophfroehlich