ros-controls/ros2_controllers #203: joint_state_broadcaster build failed

korosava, 2021-06-15:
It's similar to this [issue](https://github.com/ros-planning/moveit2/issues/462), except c++-8 replaced with c++7 here. I hope this info will be helpfull for debugging.

commits, that works for me:
```git
fake_joint             - a96af50 (HEAD -> foxy, origin/foxy) Fix the spin rate (#3)
geo shapes             - ebfc30d (HEAD -> ros2) [fix] cmake install command (#164)
moveit2                - ea62dbc6d (HEAD -> main, origin/main, origin/HEAD) MTC compatibility fixes (#323)
moveit_msgs            - 51c9a51 (HEAD -> ros2, tag: 2.0.0) 2.0.0
moveit_resources       - e512345 (HEAD -> ros2, tag: 2.0.0) 2.0.0
ros2_control           - 2c3aad1 (HEAD) New version of component parser (#127)
ros2_controllers       - 8587e07 (HEAD) [Ready] Port over interpolation formulae, abort if goals tolerance violated (#62)
srdfdom                - f77799c (HEAD -> ros2, tag: 2.0.0) 2.0.0
warehouse_ros          - d5bd4cc (HEAD -> ros2, tag: 2.0.0) 2.0.0
warehouse_ros_mongo    - 3a69993 (HEAD -> ros2, tag: 2.0.0) 2.0.0
```

last commit (that fails to build):
```git
geo_shapes             - da52ff8 (HEAD -> ros2, tag: 2.0.2, origin/ros2) 2.0.2
moveit2                - c029942db (HEAD -> main, origin/main, origin/HEAD) Temporarily disable flaky OMPL test (#495)
moveit_msgs            - 53b8ddf (HEAD -> ros2, tag: 2.0.1, origin/ros2) 2.0.1
moveit_resources       - 8d245f8 (HEAD -> ros2, origin/ros2) Add Galactic CI (#80)
ros2_control           - e980718 (HEAD -> master, origin/master, origin/HEAD) Remove forgoten debug output (#439)
ros2_controllers       - df5a5fa (HEAD -> master, origin/master, origin/HEAD) Fix rolling build (#200)
srdfdom                - ed8e5ed (HEAD -> ros2, origin/ros2) Add Galactic CI (#87)
warehouse_ros          - ec9fb64 (HEAD -> ros2, origin/ros2) Fix building on windows (#73)
warehouse_ros_mongo    - 0d0efa5 (HEAD -> ros2, origin/ros2) Add rolling CI (#56)
```

Used build instructions: 
``` bash
source /opt/ros/foxy/setup.bash && \
mkdir -p moveit2_ws/src && \
cd moveit2_ws/src && \
wget https://raw.githubusercontent.com/ros-planning/moveit2/main/moveit2.repos && \
vcs import < moveit2.repos && \
rosdep update && \
rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y && \
cd .. && \
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
```
Result:
``` c++
Starting >>> controller_manager_msgs
Starting >>> ros2_control_test_assets
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_pr2_description
Starting >>> warehouse_ros
Starting >>> geometric_shapes
Starting >>> moveit_msgs
Finished <<< ros2_control_test_assets [0.55s]                                                                                                                                                                
Finished <<< moveit_resources_panda_description [0.55s]
Finished <<< moveit_resources_fanuc_description [0.58s]                                                                                                                                 
Starting >>> hardware_interface
Starting >>> moveit_resources_fanuc_moveit_config
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_pr2_description [0.63s]
Starting >>> srdfdom
Finished <<< warehouse_ros [0.65s]                                                                                                                                                                            
Starting >>> warehouse_ros_mongo
Finished <<< geometric_shapes [0.99s]                                                                                                                                                     
Finished <<< moveit_resources_fanuc_moveit_config [0.61s]                                                                                                                                                   
Finished <<< moveit_resources_panda_moveit_config [0.61s]
Starting >>> moveit_resources
Finished <<< hardware_interface [0.69s]
Finished <<< srdfdom [0.64s]                                                                                                                                                                       
Starting >>> controller_interface
Starting >>> transmission_interface
Finished <<< controller_manager_msgs [1.37s]
[... 78 lines cut ...]
                 from /home/dalek/development/deps/moveit_ws/src/ros2_controllers/joint_state_broadcaster/test/test_joint_state_broadcaster.cpp:18:
/usr/include/c++/7/bits/stl_construct.h: In instantiation of ‘void std::_Construct(_T1*, _Args&& ...) [with _T1 = const char* const; _Args = {const char* const&}]’:
/usr/include/c++/7/bits/stl_uninitialized.h:83:18:   required from ‘static _ForwardIterator std::__uninitialized_copy<_TrivialValueTypes>::__uninit_copy(_InputIterator, _InputIterator, _ForwardIterator) [with _InputIterator = __gnu_cxx::__normal_iterator<const char* const*, std::vector<const char* const, std::allocator<const char* const> > >; _ForwardIterator = const char* const*; bool _TrivialValueTypes = false]’
/usr/include/c++/7/bits/stl_uninitialized.h:134:15:   required from ‘_ForwardIterator std::uninitialized_copy(_InputIterator, _InputIterator, _ForwardIterator) [with _InputIterator = __gnu_cxx::__normal_iterator<const char* const*, std::vector<const char* const, std::allocator<const char* const> > >; _ForwardIterator = const char* const*]’
/usr/include/c++/7/bits/stl_uninitialized.h:289:37:   required from ‘_ForwardIterator std::__uninitialized_copy_a(_InputIterator, _InputIterator, _ForwardIterator, std::allocator<_Tp>&) [with _InputIterator = __gnu_cxx::__normal_iterator<const char* const*, std::vector<const char* const, std::allocator<const char* const> > >; _ForwardIterator = const char* const*; _Tp = const char* const]’
/usr/include/c++/7/bits/stl_vector.h:331:31:   required from ‘std::vector<_Tp, _Alloc>::vector(const std::vector<_Tp, _Alloc>&) [with _Tp = const char* const; _Alloc = std::allocator<const char* const>]’
/opt/ros/foxy/src/gmock_vendor/include/gmock/gmock-matchers.h:3152:7:   required from here
/usr/include/c++/7/bits/stl_construct.h:75:13: error: invalid static_cast from type ‘const char* const*’ to type ‘void*’
     { ::new(static_cast<void*>(__p)) _T1(std::forward<_Args>(__args)...); }
             ^~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/test_joint_state_broadcaster.dir/build.make:82: CMakeFiles/test_joint_state_broadcaster.dir/test/test_joint_state_broadcaster.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:255: CMakeFiles/test_joint_state_broadcaster.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
---
Failed   <<< joint_state_broadcaster [11.3s, exited with code 2]
Aborted  <<< effort_controllers [12.6s]                                                                                                                                  
Aborted  <<< position_controllers [24.0s]                                                                                         
Aborted  <<< velocity_controllers [26.3s]                                    

Summary: 23 packages finished [29.7s]
  1 package failed: joint_state_broadcaster
  3 packages aborted: effort_controllers position_controllers velocity_controllers
  1 package had stderr output: joint_state_broadcaster
  2 packages not processed
```

bmagyar, 2021-06-15:
The package builds fine on our various CI builds and the ros build farm. The problem is likely your setup. I suggest installing binaries for foxy,saves you a lot of time.

korosava, 2021-06-15:
There is only [source build instructions](https://moveit.ros.org/install-moveit2/source/) for moveit2, where are the binaries?
I used g++ 7.5 and cmake 3.19.2 for building.

bmagyar, 2021-06-15:
I believe I have the default installation

```
$ g++ --version
g++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
```
and
```
$ cmake --version 
cmake version 3.16.3
```
for binaries follow the ROS2 Foxy guide:
https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Install-Binary

korosava, 2021-06-16:
Problem was gcc, g++ version, all works good with 9.3.0

closed 2021-06-16 by korosava