ros-controls/ros2_controllers #210: Closed loop diff_drive_controller

jonatanolofsson, 2021-07-06:
In gazebo, with varying friction parameters et.c., I can't for the life of me get a consistent behavior of the configuration of the diff_drive_controller. In fact, I struggle to get consistency between 1) reported wheel odometry 2) IMU measurements 3) Turn radius calculations from desired linear and angular rates.

The problems seem to come down to that the actual wheel speed is inconsistent with the requested wheel speed due to friction, slip et.c. The solution to this seems to be to close the loop (when open_loop=false) so that the requested wheel speed is attained. My thoughts was to add a PID to either left/right or linear/angular commands. Or am I missing something? Am I using it wrong? I no-one else experiencing these issues?

In the screenshot below, I hand-tuned the wheel multiplier to give a reasonable IMU response at 0.3 rad/s, but when issuing 0.15rad/s it's waay off.
[image]

Also - what's up with the multiplier parameters? I can't make any physical sense of them and by their performance they (at least wheel_separator....) seems as though it varies with angular speed. Are they just there in lieu of a PID, to "compensate" for errors?

shonigmann, 2021-07-06:
Another potential source of discrepancy, beyond using open-loop commands, is that gazebo_ros2_control's position and velocity interfaces act as [perfect command followers](https://github.com/ros-simulation/gazebo_ros2_control/blob/956e7706338fd4fd1a2ef733ad54740d07e9a67a/gazebo_ros2_control/src/gazebo_system.cpp#L303-L310), with infinite accelerations and discontinuous behavior (e.g. if you set a wheel velocity, the wheel will be at that velocity on the next time step). I wouldn't be surprised if that was having an impact on the odometry. that said, I agree that closing the loop on the controller would probably help too. 

there is an [effort-interface based position controller in the works](https://github.com/ros-controls/ros2_controllers/blob/effort_group_position_controller/effort_controllers/src/joint_group_position_controller.cpp) that I believe will offer more realistic behavior in Gazebo (and could be modified to be an effort-based velocity controller), without needing to tweak arbitrary multipliers, but it is a work in progress. I've been trying to build on it myself and have uncovered a number of other issues along the way, so progress has been slower than expected.

Timple, 2021-07-09:
Would it be enough if the velocity loops around the wheels where closed?

In that case the diff_drive_controller can remain open-loop again. As then the wheels turn as expected.

destogl, 2021-07-09:
> Would it be enough if the velocity loops around the wheels where closed?

I think this is how most people are using it.

cangozpi, 2023-03-22:
Hİ, I have the same issues as were initially described. Is there an already existing implementation/configuration for ros2 control's differential_drive that I can use to fix these issues ?

christophfroehlich, 2025-03-12:
https://github.com/ros-controls/ros2_control_demos has now a chained controller example with closed-loop PIDs for wheel velocity

closed 2025-03-12 by christophfroehlich