ros-controls/ros2_controllers #212: Undeclared parameters not available to controllers, despite `allow_undeclared_parameters(true)`

shonigmann, 2021-07-06:
Controllers do not seem to have access to undeclared parameters like I would expect them to. All of the ros2_controllers I have looked at [extend ros2_control's controller_interface.cpp](https://github.com/ros-controls/ros2_controllers/blob/b3300d6c0a9824c064e7d0b9f3dcf3a6ae45574b/forward_command_controller/src/forward_command_controller.cpp#L40) which explicitly sets [`allow_undeclared_parameters(true)`](https://github.com/ros-controls/ros2_control/blob/949aa8c89018ba2de630b14851b25811725b94cd/controller_interface/src/controller_interface.cpp#L33-L44)

despite this, I have had no luck getting undeclared parameters from a .yaml file to actually show up as a parameter of the controller's node. This is particularly important when implementing controllers with arbitrary limits or PID gains per joint, as it will not be possible to predefine the parameter names without changing the source for each hardware implementation. 

When using a global namespace (`/**`), the undeclared parameters show up under the `controller_manager` node only

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - ros2 foxy binaries; installed `ros2_control`, `ros2_controllers`, and `ros2_control_demos` from source in a clean workspace.
- Version or commit hash:
  - `ros2_control`: https://github.com/ros-controls/ros2_control/commit/949aa8c89018ba2de630b14851b25811725b94cd
  - `ros2_controllers`: https://github.com/ros-controls/ros2_controllers/commit/b3300d6c0a9824c064e7d0b9f3dcf3a6ae45574b
  - `ros2_control_demos`: https://github.com/shonigmann/ros2_control_demos/commit/7db8ec39c724133714735ef77deab49a92080e51 (ros2_control_demos master + small changes to config file and urdf for testing minimal reproduceable example)

#### Steps to reproduce issue

**Terminal 1:**
```
mkdir -p ~/test_ws/src
cd ~/test_ws/src
git clone https://github.com/ros-controls/ros2_control.git
git clone https://github.com/ros-controls/ros2_controllers.git
git clone https://github.com/shonigmann/ros2_control_demos.git
cd ~/test_ws
rosdep install --from-paths src --ignore-src -y
colcon build --symlink-install
. install/setup.bash
ros2 launch ros2_control_demo_bringup rrbot_system_multi_interface.launch.py
```

**Terminal 2:**
```
ros2 param list
```

#### Expected behavior
the fork of ros2_control_demos linked above includes additional, undeclared parameters in `rrbot_multi_interface_forward_controllers.yaml`. It is expected that these undeclared parameters are declared when the controller (e.g. `forward_position_controller`) is initialized. 

Listing parameters using `ros2 param list` is expected to show something to the effect of:
```
/controller_manager:
  forward_acceleration_controller.type
  forward_illegal1_controller.type
  forward_illegal2_controller.type
  forward_position_controller.type
  forward_velocity_controller.type
  joint_state_broadcaster.type
  robot_description
  test_1
  test_2
  test_gains.test_3.d
  test_gains.test_3.i
  test_gains.test_3.p
  test_limits.test_4.has_position_limits
  test_limits.test_4.x_max
  test_limits.test_4.x_min
  update_rate
  use_sim_time
/forward_position_controller:
  joints
  gains.joint1.p
  gains.joint1.i
  gains.joint1.d
  limits.joint1.has_position_limits
  limits.joint1.x_min
  limits.joint1.x_max
  use_sim_time
  test_1
  test_2
  test_gains.test_3.d
  test_gains.test_3.i
  test_gains.test_3.p
  test_limits.test_4.has_position_limits
  test_limits.test_4.x_max
  test_limits.test_4.x_min
... etc...
```

#### Actual behavior
Undeclared parameters defined in a controller's namespace are not listed. Undeclared parameters defined on the `/**` namespace are only listed under the `controller_manager` node:
```
ros2 param list
/controller_manager:
  forward_acceleration_controller.type
  forward_illegal1_controller.type
  forward_illegal2_controller.type
  forward_position_controller.type
  forward_velocity_controller.type
  joint_state_broadcaster.type
  robot_description
  test_1
  test_2
  test_gains.test_3.d
  test_gains.test_3.i
  test_gains.test_3.p
  test_limits.test_4.has_position_limits
  test_limits.test_4.x_max
  test_limits.test_4.x_min
  update_rate
  use_sim_time
/forward_position_controller:
  joints
  use_sim_time
/joint_state_broadcaster:
  use_sim_time
/robot_state_publisher:
  ignore_timestamp
  publish_frequency
  robot_description
  use_sim_time
  use_tf_static
/rviz2:
  use_sim_time
```
#### Additional information
Existing workarounds that I am aware of include:

- explicitly defining parameters, such as in [JointTrajectoryController](https://github.com/ros-controls/ros2_controllers/blob/b3300d6c0a9824c064e7d0b9f3dcf3a6ae45574b/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L63-L71); this is not ideal for generic controllers as stated above
- explicitly define parameter arrays (e.g. `node_->declare_parameter<std::vector<std::string>>("gains.p", p_gains_);`), and ensure that the indices with each parameter match the `joints` parameter array. This is perhaps the most direct approach, but feels clunky and very prone to user error (e.g. changing the wrong index of a limits array could be dangerous)
- setting undefined parameters in the .yaml config file for the _controller manager only_, and calling the `/get_parameters` service to query for parameters. This feels a bit hacky, and would require lengthy parameter names to avoid clashing when different controllers are loaded (e.g. if multiple controllers have different gains per joint). 

but none of these are ideal

jordan-palacios, 2021-07-07:
Can you try adding `automatically_declare_parameters_from_overrides(true)` to the `node_options`? I think that should allow you to completely define parameters from yaml files.

shonigmann, 2021-07-07:
Thanks @jordan-palacios  - I'd tried that earlier and had moved away due to resultant errors (which I'll include below for reference). Now that I'm a bit more familiar with the code, the errors made more sense and were easily fixable! 

For reference to anyone else interested, adding `automatically_declare_parameters_from_overrides(true)` to the `node_options` e.g. in `forward_position_controller.cpp`:
```
ForwardCommandController::init(const std::string & controller_name)
{
  auto node_options_ = rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(true);  // <-- add this
  auto ret = ControllerInterface::init(controller_name, node_options_);  // <-- add node_options_ to ControllerInterface
  if (ret != controller_interface::return_type::OK) {
    return ret;
  }
  ...
```
will result in:
```
[ros2_control_node-1] [INFO] [1625674883.611064919] [controller_manager]: Loading controller 'forward_position_controller'
[ros2_control_node-1] Exception thrown during init stage with message: parameter 'joints' has already been declared 
[ros2_control_node-1] [ERROR] [1625674883.614263195] [controller_manager]: Could not initialize the controller named 'forward_position_controller'
[ros2_control_node-1] [INFO] [1625674883.615902617] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ERROR] [spawner.py-5]: process has died [pid 230621, exit code 1
```
Because `forward_command_controller.cpp` returns an `ERROR` if `declare_parameter` fails
```
  try {
    auto node = get_node();
    node->declare_parameter<std::vector<std::string>>("joints", std::vector<std::string>());

    node->declare_parameter<std::string>("interface_name", "");
  } catch (const std::exception & e) {
    fprintf(stderr, "Exception thrown during init stage with message: %s \n", e.what());
    return controller_interface::return_type::ERROR;  // <-- causes controller to quit
  }
```
Commenting out that return statement or simply commenting out the whole try-catch block lets things work as expected!

closed 2021-07-07 by shonigmann