ros-controls/ros2_controllers #214: JTC Open-loop Control Mode: Should `last_commanded_state_` here be reduced to existing state interface from the hardware? Or is it OK to internally manage more control-states then can be received from hardware?

destogl, 2021-07-12:
Should `last_commanded_state_` here be reduced to existing state interface from the hardware? Or is it OK to internally manage more control-states then can be received from hardware?

_Originally posted by @destogl in https://github.com/ros-controls/ros2_controllers/pull/189#r642045259_

christophfroehlich, 2023-09-09:
@destogl could you please explain what you meant here? I'd be happy to implement that.

You mean that it should not set the `point_before_trajectory_msg` from the commands, if there is no state interface with identical hardware interface for the joint? What would be the alternative? Just use the first entry of the trajectory message?

destogl, 2023-09-20:
Not worth to take care of it.

closed 2023-09-20 by destogl