ros-controls/ros2_controllers #221: JointGroupPositionController fails to initialize on Rolling due to rclcpp "undeclare_parameter" API change

schornakj, 2021-08-11:
There have already been [some fixes to other controllers](https://github.com/ros-controls/ros2_controllers/pull/199) for the rclcpp parameter API changes. Currently (as of 0.4.1) the `JointGroupPositionController` fails to initialize because it tries to undeclare a statically-typed parameter at [these lines](https://github.com/ros-controls/ros2_controllers/blob/6a53981982eeac1822bfca8e9d5e40156870c0b1/position_controllers/src/joint_group_position_controller.cpp#L41-L47), which produces an error:

```
[ros2_control_node-1] [INFO] [1628693935.184140389] [controller_manager]: Loading controller 'position_controller'
[ros2_control_node-1] Exception thrown during init stage with message: parameter 'interface_name' has invalid type: cannot undeclare an statically typed parameter 
[ros2_control_node-1] [ERROR] [1628693935.194137855] [controller_manager]: Could not initialize the controller named 'position_controller'
```

I wanted to get some additional context and input before I try to fix this myself: why is it necessary to undeclare the `interface_name` parameter declared by `ForwardCommandController`, and does anyone else already have a fix in mind?

gc-robotics, 2021-08-14:
I am facing the same issue.

Exception thrown from the following line:
`    // undeclare interface parameter used in the general forward_command_controller
    get_node()->undeclare_parameter("interface_name");`

gc-robotics, 2021-08-14:
Have just seen the PR. Will wait for the approval. Thanks @schornakj .

PR #222 (2021-08-20): Delete failing parameter undeclare in JointGroupPositionController
This is one way to fix #221 and allow loading the JointGroupPositionController plugin on Rolling.

If I understand correctly, the JointGroupPositionController just sets the value of `interface_name_` to `hardware_interface::HW_IF_POSITION` instead of reading it from a parameter, so it shouldn't make a functional difference whether the parameter is explicitly undeclared or simply not read.
closed 2021-08-20 by bmagyar