ros-controls/ros2_controllers #242: Differential drive acceleration limiting not working properly

kallaspriit, 2021-09-20:
The acceleration limiting feature for differential drive controller does not seem to be working properly.

I added some logging and it seems the acceleration limiting is only applied for one frame. For the next frame, the base `v0` seems to be taken from the command not the limited value so it does not see acceleration any more.

Tested on `foxy` with `linear.x.has_velocity_limits: true` and `linear.x.max_acceleration: 0.1`.

I'm requesting instantaneous speed from 0 to 1 m/s and I'm expecting to see gradual rise in speed but in reality I'm seeing acceleration limiting for only one update (0 > 0.001015 > 1.0 m/s).

```c++
double SpeedLimiter::limit_acceleration(double & v, double v0, double dt)
{
  const double tmp = v;

  if (has_acceleration_limits_)
  {
    const double dv_min = min_acceleration_ * dt;
    const double dv_max = max_acceleration_ * dt;

    const double dv = rcppmath::clamp(v - v0, dv_min, dv_max);

    v = v0 + dv;

    // Acc: dv: 0.000000, v0: 0.000000, before: 0.000000, after: 0.000000
    // Acc: dv: 0.001015, v0: 0.000000, before: 1.000000, after: 0.001015
    // Acc: dv: 0.000000, v0: 1.000000, before: 1.000000, after: 1.000000
    RCLCPP_INFO(rclcpp::get_logger("SpeedLimiter"), "Acc: dv: %f, v0: %f, before: %f, after: %f", dv, v0, tmp, v);
  }

  return tmp != 0.0 ? v / tmp : 1.0;
}
```

Am I understanding something wrong?

bmagyar, 2021-09-24:
Please add a test that reproduces this issue so we can introduce a fix for it.

bmagyar, 2021-10-09:
We'd really prefer if you could do the fix against the latest master branch using Galactic. I'm happy to backport to Foxy but we prefer not to make "forward porting" a thing :) 

Indeed Foxy will receive non-breaking updates from upstream as long as you guys want them and as long as I remember to do them (or you keep reminding me :))

joshnewans, 2021-10-09:
Sure thing, I'll do it on master and spin up a galactic docker container for a test to make sure it works.

As far as a proper test goes, that's not something I've got experience with, especially not within the context of ROS controllers - but I guess this is as good an opportunity as any to learn!
If you can point me in the right direction I'll see what I can do. :)

joshnewans, 2021-10-20:
Yep, good to close.

kallaspriit, 2021-10-21:
Thanks, can we get this in LTS Foxy branch too?

kallaspriit, 2021-10-30:
Built the foxy version from source and can confirm it working nicely now on a physical robot :)

closed 2021-10-30 by kallaspriit

ketill1, 2024-05-10:
Is this fix also implemented on humble and above?

christophfroehlich, 2024-05-10:
Yes, the fix of #252 is included starting from foxy up to every newer distro.

ketill1, 2024-05-12:
Thank you @christophfroehlich  for the fast answer. I'm still experiencing this exact issue on a physical robot. I have installed ros-humble-ros2-control and ros-humble-ros2-controllers with apt. Any Idea of what the issue can be or where to find some help?