ros-controls/ros2_controllers #243: Does not build on galactic.

tylerjw, 2021-09-20:
**Describe the bug**
As ros2_control/ros2_controllers are not released for Galactic the moveit2 CI system depends on source builds of this repo.  The latest changes merged here broke our CI builds for galactic.  This will be helped after ros2_control is released for galactic but in the meantime maybe these repos need galactic CI to pass so they keep building on galactic in the master branch?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/ros-planning/moveit2/actions
3. Scroll down to 'any recent PR' and open the Galactic CI jobs
4. See error

**Expected behavior**
Master branch builds on Galactic.

bmagyar, 2021-09-24:
Unfortunate but we can't be doing braking changes before the release *while* maintaining a full build. We typically try to minimize the window of ros2_control and ros2_controllers being broken together but rest assured, we don't merge anything on ros2_control that would not have the ros2_controllers PR open and ready (if required).

For now, please confirm that the issues are gone and close this?

tylerjw, 2021-09-24:
I tried yesterday and while it did build some behavior changed I haven't figured out so we won't be switching back until I understand how to update our tests.

bmagyar, 2021-09-26:
Here's a candidate: https://github.com/ros2/rviz/pull/746