ros-controls/ros2_controllers #247: Make resetting RealtimeBuffer easier by introducing a reset() method

bmagyar, 2021-09-25:
## Background

It turns out, some of our controllers forgot to reset their buffers when deactivating/reactivating!

We've implemented a quick fix for it in https://github.com/ros-controls/ros2_controllers/pull/246 (see the `Files Changed` tab) but it could be improved.

Resetting the buffer currently looks something like this:
```
  rt_command_ptr_ = realtime_tools::RealtimeBuffer<std::shared_ptr<CmdType>>(nullptr);
```
Oh no, that's long and ugly! :scream: 

We'd rather like to see
```
  rt_command_ptr_->reset();
```

We need to change the `RealtimeBuffer` API for this and extend it with a `reset()` function that does the above.
Go to https://github.com/ros-controls/realtime_tools/tree/foxy-devel (note the choice of the branch) and add said `reset()` function to the code. Open a Pull Request against the `foxy-devel` branch and once merged, replace that long and ugly line in this repository to make use of this new function!

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_controllers/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/foxy/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://ros-controls.github.io/control.ros.org/getting_started.html#compiling).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_controllers`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_controllers` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://ros-controls.github.io/control.ros.org/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/foxy/Tutorials.html)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

radekrudak, 2021-09-27:
Can reset function take pointer to o ject it belong to as argument ?

bailaC, 2021-10-12:
@bmagyar , @destogl you can assign this to me. I will work on this.

bmagyar, 2021-11-18:
This can proceed now further, the PR on realtime_tools was merged and released and synched to Foxy, Galactic and Rolling

christophfroehlich, 2025-03-14:
Partially fixed with #283, but got introduced again:
https://github.com/search?q=repo%3Aros-controls%2Fros2_controllers+%22%3D+realtime_tools%3A%3ARealtimeBuffer%3Cstd%3A%3Ashared_ptr%22&type=code

christophfroehlich, 2025-07-26:
We moved from RealtimeBuffer to RealtimeThreadSafeBox with #1721. The current pattern is a try_set to a default message in on_activate and on_deactivate (both are in the RT thread!)
https://github.com/ros-controls/ros2_controllers/blame/b21a7e1d296f5176538f874693de7472cda7fbf7/forward_command_controller/src/forward_controllers_base.cpp#L131-L135
and
https://github.com/ros-controls/ros2_controllers/blob/b21a7e1d296f5176538f874693de7472cda7fbf7/forward_command_controller/src/forward_controllers_base.cpp#L144-L146