ros-controls/ros2_controllers #248: joint_state_broadcaster in rolling fails to build on rhel targets

nuclearsandwich, 2021-09-29:
The package joint_state_broadcaster in rolling has been detected as not building on all platforms in the buildfile rhel. The release manager for rolling will consider disabling this build if it continues to fail to build.
- jenkins_urls:
   - [rhel 8 x86_64](https://build.ros2.org/view/Rbin_rhel_el864/job/Rbin_rhel_el864__joint_state_broadcaster__rhel_8_x86_64__binary)
This is being filed because this package is about to be blacklisted. If this ticket is resolved please review whether it can be removed from the blacklist that should cross reference here.

bmagyar, 2021-09-30:
This only affect the Red Hat build, right?

destogl, 2021-09-30:
@nuclearsandwich How are you developing other packages for RHEL? Is there a docker, or some GH-Action or something to test this before we trigger release?

bmagyar, 2021-10-30:
Thanks guys for the support, we will try to add a rolling RHEL job to our master branches for keeping an eye on it. 

For me, @cottsay 's comment re. "non-Ubuntu fixes" is the jackpot. We had issues with Windows and Mac builds but those can only be reproduced sporadically by the few users on those platforms while this actually brings us a reproducible, CI-compatible alternative.

jaron-l, 2021-11-29:
@cottsay, if you are working to add RHEL support, is OSRF planning on publishing some sort of official ros docker image for a RHEL variant? I wrote my own to use in CI based on the installation instructions but wasn't sure if I need to set up a pipeline to keep it updated or if you already have that somewhere. It's easy for me to do I just wanted to know if it's best put somewhere more public than my personal Docker hub account.

I ask because if ros-controls is going to add this to our pipeline, we would need a Docker image to based our industrial_ci pipeline on.

cottsay, 2021-12-02:
> is OSRF planning on publishing some sort of official ros docker image for a RHEL variant?

We've discussed it. A non-trivial amount of work in [osrf/docker_templates](https://github.com/osrf/docker_templates) will be necessary, and I'm not sure it's going to land any time soon.

jaron-l, 2022-08-17:
Closing this issue as the CI now runs and builds correctly on rolling

closed 2022-08-17 by jaron-l