ros-controls/ros2_controllers #249: JointTrajectoryController ignores tolerances sent via action client

livanov93, 2021-10-07:
**Would be nice to have**
`JointTrajectoryController` ignores `goal_time_tolerance` set via [FollowJointTrajectory.Goal.goal_time_tolerance](http://docs.ros.org/en/api/control_msgs/html/action/FollowJointTrajectory.html) together with `path_tolerance` and `goal_tolerance` .

**Potential ROS1 inspiration**
Implement something like a ROS1 version [`updateSegmentTolerances`](https://github.com/ros-controls/ros_controllers/blob/aa9c32b8ea8f83cb578cde82619f78e720fef89c/joint_trajectory_controller/include/joint_trajectory_controller/tolerances.h#L232) which is called every time new goal is received through [`updateTrajectoryCommand`](https://github.com/ros-controls/ros_controllers/blob/aa9c32b8ea8f83cb578cde82619f78e720fef89c/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L604) or whenever [new trajectory is received](https://github.com/ros-controls/ros_controllers/blob/aa9c32b8ea8f83cb578cde82619f78e720fef89c/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L169).
The parameter propagation is continued on the call of [`initJointTrajectory`](https://github.com/ros-controls/ros_controllers/blob/aa9c32b8ea8f83cb578cde82619f78e720fef89c/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L532) which in the end uses `updateSegmentTolerances` [here](https://github.com/ros-controls/ros_controllers/blob/aa9c32b8ea8f83cb578cde82619f78e720fef89c/joint_trajectory_controller/include/joint_trajectory_controller/init_joint_trajectory.h#L318)

This is needed to be able to overwrite constraints that are set via parameter server on the jtc start.

lukicdarkoo, 2021-12-25:
It seems that the `constraints.goal_time` parameter is also ignored (at least for state position + control position).

I am getting `Goal finished with status: SUCCEEDED` even if the joint hasn't reached the position.
```
ros2 action send_goal /ur_joint_trajectory_controller/follow_joint_trajectory control_msgs/action/FollowJointTrajectory "{
>   trajectory: {
>     joint_names: [shoulder_pan_joint],
>     points: [
>       { positions: [3.02], time_from_start: { sec: 0, nanosec: 100 } },
>     ]
>   },
>   goal_tolerance: [
>     { name: shoulder_pan_joint, position: 0.00000001 },
>   ]
> }"
Waiting for an action server to become available...
Sending goal:
     trajectory:
  header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: ''
  joint_names:
  - shoulder_pan_joint
  points:
  - positions:
    - 3.02
    velocities: []
    accelerations: []
    effort: []
    time_from_start:
      sec: 0
      nanosec: 100
path_tolerance: []
goal_tolerance:
- name: shoulder_pan_joint
  position: 1.0e-08
  velocity: 0.0
  acceleration: 0.0
goal_time_tolerance:
  sec: 0
  nanosec: 0

Goal accepted with ID: 816cd999bcca4225b6f7c027cb64998e

Result:
    error_code: 0
error_string: ''

Goal finished with status: SUCCEEDED
```

livanov93, 2022-01-08:
This issue should motivate to implement an option to overwrite tolerances that are sent via action goal. This is not implemented yet. For now, if you want to enforce constraints on the state (during trajectory) and on the final (goal) configuration you should set parameters like shown [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/a0cfee50ae7b9318ad705d9634716a76cdb131a7/ur_bringup/config/ur_controllers.yaml#L65-L73) in UR example.

`constraints.goal_time` just makes sure the open-loop control has correct timing on the end of trajectory (e.g. trajectory is executed within the specified time). If it is set to 0, it is ignored. For details see [here](https://github.com/ros-controls/ros2_controllers/blob/9733da06dad536209d514200562ddbb744938c33/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L284-L302)

PR #716 (2024-07-03): [JTC] Process tolerances sent with action goal
From [here](https://github.com/ros-controls/ros2_controllers/issues/249#issue-1019945122):

> JointTrajectoryController ignores goal_time_tolerance set via [FollowJointTrajectory.Goal.goal_time_tolerance](http://docs.ros.org/en/api/control_msgs/html/action/FollowJointTrajectory.html) together with path_tolerance and goal_tolerance .

This PR proposes a way how to process the tolerances: The tolerances from one action goal are not saved for the next one, but the default ones will be used if no tolerances are set with the following action goals.

(T [... cut]
PR #1190 (2024-07-15): [JTC] Process tolerances sent with action goal (backport #716)
From [here](https://github.com/ros-controls/ros2_controllers/issues/249#issue-1019945122):

> JointTrajectoryController ignores goal_time_tolerance set via [FollowJointTrajectory.Goal.goal_time_tolerance](http://docs.ros.org/en/api/control_msgs/html/action/FollowJointTrajectory.html) together with path_tolerance and goal_tolerance .

This PR proposes a way how to process the tolerances: The tolerances from one action goal are not saved for the next one, but the default ones will be used if no tolerances are set with the following a [... cut]
PR #1189 (2024-07-15): [JTC] Process tolerances sent with action goal (backport #716)
From [here](https://github.com/ros-controls/ros2_controllers/issues/249#issue-1019945122):

> JointTrajectoryController ignores goal_time_tolerance set via [FollowJointTrajectory.Goal.goal_time_tolerance](http://docs.ros.org/en/api/control_msgs/html/action/FollowJointTrajectory.html) together with path_tolerance and goal_tolerance .

This PR proposes a way how to process the tolerances: The tolerances from one action goal are not saved for the next one, but the default ones will be used if no tolerances are set with the following a [... cut]
closed 2024-07-03 by bmagyar