ros-controls/ros2_controllers #251: Create test for diff_drive_controller motion limits

joshnewans, 2021-10-10:
Issue #242 revealed a bug with the diff_drive_controller's velocity/acceleration/jerk limit code.

Although a PR has been made to fix this, a test should be created to prevent this in future.

roncapat, 2021-11-06:
I may contribute to this since I'm working on other tests. 
Any preferences on the behaviour to test? I mean, a sequence of steps and checks, a numeric input/ expected output example. Otherwise I will try to define a good test on my own in the following weeks :)

christophfroehlich, 2025-04-22:
The speed limiter has moved to the control_toolbox, and got some tests
https://github.com/ros-controls/control_toolbox/pull/212
https://github.com/ros-controls/control_toolbox/pull/237

closed 2025-04-22 by christophfroehlich