ros-controls/ros2_controllers #262: Remapping cmd_vel topic

kallaspriit, 2021-10-29:
Hey,

Is there some way to remap the velocity command topic to use `/cmd_vel` instead of `/diff_drive_controller/cmd_vel_unstamped` (or `/diff_drive_controller/cmd_vel` when stamped)?

I'm trying to use the diff drive controller with [Nav2](https://navigation.ros.org/getting_started/index.html) but I can't find a way to either make the diff drive controller to listen to `/cmd_vel` nor make Nav2 publish `/diff_drive_controller/cmd_vel_unstamped`.

This is how I'm launching diff drive controller as part of my launch python script but I guess because it's using `spawner.py` then the usual `remappings` option does not seem to work.

```python
# setup diff drive controller
diff_drive_controller_node = Node(
    package="controller_manager",
    executable="spawner.py",
    arguments=["diff_drive_controller"],
    output="screen",
    # this does not really work
    remappings={
        ('/diff_drive_controller/cmd_vel_unstamped', '/cmd_vel')
    },
)
```

Any recommendations how to get diff drive to listen to `/cmd_vel` or make Nav2 output `/diff_drive_controller/cmd_vel_unstamped`?

kallaspriit, 2021-10-31:
When I clone the `ros2_controllers` repo in my workspace and just replace values of `DEFAULT_COMMAND_TOPIC` and `DEFAULT_COMMAND_UNSTAMPED_TOPIC` in `diff_drive_controller.cpp`, everything works as expected.

[image]

But this is not great as I can't just update packages and have to build this package on the robot.

Would it be possible to make the topic names configurable?

destogl, 2021-10-31:
I don't understand exactly the problem. In your first post, it appears to me that you are remapping topic on the wrong node. Controller's nodes are created from `ros2_control_node` and not from `spawner.py`. The latter is destroyed shortly after controller is started.

I am successfully remapping topic in my applications when doing this on `ros2_control_node`. Can you try this?

Other than that, you can simply remap “other side” of the topic where is published. Here `Nav2`.

destogl, 2021-10-31:
@kallaspriit I extended our examples in ros-controls/ros2_control_demos#143 with topic remapping. Please check `diffbot.launch.py` file and let me know if example is working for you.

kallaspriit, 2021-10-31:
Thanks, this works well, indeed tried to do the mapping in the wrong place, this makes sense :)

closed 2021-10-31 by kallaspriit

xibeisiber, 2023-02-27:
the remap of cmd_vel on diffbot.launch.py seems not working...

XinyuWang1221, 2023-04-18:
@destogl HI. Is this way also working for the tricycle controller?  I've tried, but it doesn't work. like here:

        Node(
           package='robot_state_publisher',
           executable='robot_state_publisher',
           name='robot_state_publisher',
           output='screen',
           parameters = [{'use_sim_time':True,'robot_description':
            desc,}],
           remappings= [("/tricycle_controller/cmd_vel","/cmd_vel"),
                         ("/tricycle_controller/odom","/odom")]
           #arguments = [urdf]
            ),

spok7, 2024-05-03:
For those confused on how to do this, it should be done by remapping the ros2_control_node like so:
```python
Node(
    package="controller_manager",
    executable="ros2_control_node",
    remappings=[
        ("/diff_drive_controller/cmd_vel_unstamped", "/cmd_vel"),
    ]
)
```
If you named your controller anything else, make sure to change `diff_drive_controller` accordingly. Here's an example launch description:
```python
def generate_launch_description():

    urdf_file = get_package_share_path("robot_base")
    urdf_file /= "urdf"
    urdf_file /= "robot.urdf.xacro"
    
    urdf_node = Node(
        package="robot_state_publisher",
        executable="robot_state_publisher",
        parameters=[{
            "robot_description": ParameterValue(
                Command(["xacro ", str(urdf_file)]), value_type=str
            )
        }]
    )
    
    motor_control_node = Node(
        package="controller_manager",
        executable="ros2_control_node",
        parameters=[PathJoinSubstitution([
            FindPackageShare("robot_base"),
            "config",
            "robot_base_controllers.yaml",
        ])],
        remappings=[
            ("~/robot_description", "/robot_description"),
            ("/diffbot_base_controller/cmd_vel_unstamped", "/cmd_vel"),
        ]
    )

    motor_controller_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=["diffbot_base_controller",
                   "--controller-manager", "/controller_manager"]
    )

    joint_state_broadcaster_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=["joint_state_broadcaster",
                   "--controller-manager", "/controller_manager"]
    )

    return LaunchDescription([
        nodes_to_launch_launch_arg,
        urdf_node,
        motor_control_node,
        joint_state_broadcaster_spawner,
        RegisterEventHandler(event_handler=OnProcessExit(
            target_action=joint_state_broadcaster_spawner,
            on_exit=[motor_controller_spawner]
        ))
    ])
        
```