ros-controls/ros2_controllers #264: Mobile Robotics Controllers (Ackermann, four_wheels, swerve, ...)

tomlogan501, 2021-11-08:
Hi,
I create just this topic to keep track of the transfert of those controllers from ROS1 to ROS2.
We can discuss the main changes and the benefit to do so (new architecture, cyclic, ...)

harderthan, 2021-12-14:
Hello. 

I create [ackermann-steering-controller-ros2 repository](https://github.com/harderthan/ackermann-steering-controller-ros2) and implement a migration for ros1 [ackermann_steering_controller](https://github.com/ros-controls/ros_controllers/tree/noetic-devel/ackermann_steering_controller)  

I finished only to build the migrated code. I have a todo list below.

#### TODO 
- [x] Trouble shooting to build
- [ ] Add the Odometry/Velocity Calculation on ackermann_steering_controller.cpp
- [ ] Add test codes for ackermann_steering_controller.cpp
- [ ] Update Odometry.hpp and .cpp
- [ ] Update documents (remarks, descriptions, etc.)
- [ ] Add Test Code
- [ ] PR

Advice and questions are welcome. thank you.

Timple, 2021-12-14:
Be carefull with 'just' porting from ROS1.
For ROS1 ackermann, the implementation seemed [off](https://github.com/ros-controls/ros_controllers/issues/412).

Also support for front-wheel drive seems to be missing, but that might be a new feature request.

harderthan, 2021-12-14:
> For ROS1 ackermann, the implementation seemed off.  

@Timple  Reallly thank you for the feedback! I didn't know it.  

Do you know if there was any discussion about adding an Ackermann controller? I'm not sure how I should contribute. Does it make sense to read and implement all [this design_drafts](https://github.com/ros-controls/roadmap/tree/master/design_drafts)? 

And, I couldn't find the information about 'front-wheel drive'.
Do you mean `four_wheel_steering_controller`?

bmagyar, 2021-12-17:
@harderthan please don't create a separate repository with code copied. 
The way to contribute changes and get feedback is to first fork this repo and add a new package.

For porting of controllers you don't need a design draft. I agree with @Timple *however* I think the first step should be to do the porting of that controller as-is, then you can go and improve it. Doing too many things at once will only get you into a situation where it's impossible to trace your steps back and impossible to review as well.

Here's my suggestion for a chain of PRs once you have most things working (applicable to any of the controllers above):
* Basic package structure implementing a controller that does nothing in particular, should have plugin, main header file, main cpp file, cmake properly set up, a load-controller test (see other controllers), no parameters, no subscriber or publisher
* Add all basic subscribers and publishers, doing the least amount of activity, add basic controller documentation (see other controllers)
* Add the implementation of the controller, subscriber callbacks and publishing to topics, add basic functional tests
* Expose parameters and add tests that rely on setting parameters
* Add any more tests that were not added yet

When in doubt, check the diff_drive_controller although it's not in a great shape so if you have better ideas for some things, feel free to be smarter.
Also take a look at our contribution guidelines: http://control.ros.org/contributing.html

aboarya, 2022-01-29:
Has any further work been done on this issue? I also need to implement a car-like steering controller for my current project.

tonynajjar, 2022-07-15:
@bmagyar are you getting the notifications on this PR? Could you advise on how to proceed?

https://github.com/ros-controls/ros2_controllers/pull/345

bmagyar, 2022-07-17:
As far as I'm concerned @tonynajjar your PR is the way to go. Sorry for the delay on reviewing, we've been really busy with framework features and controllers received a little less attention. We have the reviewer lottery to help with that but sometimes you get unlucky with the draw

DiuLaMaX, 2022-12-08:
@bmagyar What is the current status of migrating Ackermann_steering_controller from ROS1 to ROS2?

hishambaharom, 2022-12-12:
@bmagyar when the ackermann_steering_controller support on ros2 humble

bmagyar, 2022-12-13:
@DiuLaMaX @hishambaharom it's waiting for you guys to put the work in ;)

christophfroehlich, 2025-05-21:
Closing, as the remaining controllers are handled in #648 and #1425