ros-controls/ros2_controllers #265: Timing issue with Joint Trajectory Controller in Github CI with very slow simulations

BenjaminHug8, 2021-11-09:
**Describe the bug**
We are using `Joint Trajectory` controllers in one of our demos with robotic arms and have a system test to check the functionalities.

Everything works fine on local (`Ubuntu 20.04` with `Foxy` or `Rolling`) but when the test is run on `Github CI` (also Ubuntu with `Foxy` or `Rolling`), 15% of the time it will fail like [here](https://github.com/cyberbotics/webots_ros2/runs/4136703759?check_suite_focus=true#step:3:17579), as command are sent before the `joint trajectory` is ready (guess from the line with `[spawner.py-62] [INFO] [1636363807.141690639] [abb.spawner_abb_joint_trajectory_controller]: Configured and started abb_joint_trajectory_controller` shown after the error).

[Here](https://github.com/cyberbotics/webots_ros2/blob/fix-universal-robots-CI/webots_ros2_universal_robot/webots_ros2_universal_robot/abb_controller.py#L75) is the controller used for the abb arm, using a [custom class](https://github.com/cyberbotics/webots_ros2/blob/fix-universal-robots-CI/webots_ros2_universal_robot/webots_ros2_universal_robot/follow_joint_trajectory_client.py#L64) that should wait for the `action server` to be ready with the function `wait_for_server()`.

[Here](https://github.com/cyberbotics/webots_ros2/blob/fix-universal-robots-CI/webots_ros2_universal_robot/launch/multirobot_launch.py#L88-L94) you can find the launcher of the simulation focused on the abb arm.

My guess is that maybe the `wait_for_server()` function will return when the `joint trajectory` controller is initialized but not yet ready to receive commands. Locally the code will run fast enough for the controller to be ready to receive commands but on CI as our simulation run very slowly sometimes it will not have sufficient time. Would such a thing be possible ?

I can solve my problems by adding `time.sleep()` but imo it is not a good way to solve this.

Maybe is there another function that will check that the `trajectory controller`  is ready to receive the commands, but all I can find is to use the function `wait_for_server()` or to use `time.sleep()`.

BenjaminHug8, 2021-11-15:
Anyone has an idea about my issue?

BenjaminHug8, 2021-11-17:
The issue has been solved on our side by waiting a `JointTrajectoryControllerState` message on the `/state` topic, confirming that the `JointTrajectoryController` is ready to receive goals.

closed 2021-11-17 by BenjaminHug8

bmagyar, 2021-11-18:
Nice! We can probably fix our flaky rostests the same way!