ros-controls/ros2_controllers #271: [DiffDriveController] use "period" argument in the update method insead of calculating "dt"

destogl, 2021-11-18:
> we should have `period` available in the `update` method

_Originally posted by @destogl in https://github.com/ros-controls/ros2_controllers/pull/260#discussion_r752228051_

christophfroehlich, 2025-03-14:
Consider also the discussion in 
https://github.com/ros-controls/ros2_controllers/pull/1481

christophfroehlich, 2025-03-19:
> @christophfroehlich can you review the odometry implementation in #1535? If it looks good, I can make a PR making similar modifications to the ``diff_drive_controller``.

_Originally posted by @Amronos in https://github.com/ros-controls/ros2_controllers/issues/1394#issuecomment-2726314076_

Amronos, 2025-07-26:
Please assign this to me. I will start working on it in about a week.