ros-controls/ros2_controllers #272: Support for ros2-galactic, missing base_interface.hpp?

3mrrrx, 2021-11-18:
I might be very new to ros2 but i am trying to get some code running on galactic that was actually meant for foxy.

apparently the code  falling on with this error:

`fatal error: hardware_interface/base_interface.hpp: No such file or directory
   20 | #include <hardware_interface/base_interface.hpp>`

so my question would be if this is just/still a missing library for ros2-galactic or a fundamental change to ros2-foxy?

cheers and  thank you for the support

3mrrrx, 2021-11-18:
i also noticed this:

these the available libraries in for both galatic and foxy.

`
ros-galactic-dynamic-edt-3d ros-galactic-dynamixel-sdk-custom-interfaces-dbgsym
ros-galactic-dynamic-edt-3d-dbgsym ros-galactic-dynamixel-sdk-dbgsym
ros-galactic-dynamixel-sdk ros-galactic-dynamixel-sdk-examples
ros-galactic-dynamixel-sdk-custom-interfaces ros-galactic-dynamixel-sdk-examples-dbgsym

ros-foxy-dynamic-edt-3d ros-foxy-dynamixel-sdk
ros-foxy-dynamic-edt-3d-dbgsym ros-foxy-dynamixel-sdk-custom-interfaces
ros-foxy-dynamic-graph ros-foxy-dynamixel-sdk-custom-interfaces-dbgsym
ros-foxy-dynamic-graph-dbgsym ros-foxy-dynamixel-sdk-dbgsym
ros-foxy-dynamixel-hardware-interface ros-foxy-dynamixel-sdk-examples
ros-foxy-dynamixel-hardware-interface-dbgsym ros-foxy-dynamixel-sdk-examples-dbgsym
`

apperently ros-galactic-dynamixel-hardware-interface doesn't exist!!!

any ideas how to fix this?

bmagyar, 2021-11-29:
> i am trying to get some code running on galactic that was actually meant for foxy.

So you are porting some code over to Galactic, nice!
For everything non-trivial, we keep a section in our documentation page to help migrating between ROS2 distros with ros2_control: http://control.ros.org/ros2_control/hardware_interface/doc/hardware_components_userdoc.html?highlight=galactic#migration-from-foxy-to-galactic

For the missing packages... please contact the maintainers of those.
Also, in the future please use https://answers.ros.org for queries like this. Only submit an issue here once you confirmed it is something that needs to be fixed upstream.

closed 2021-11-29 by bmagyar

Timple, 2021-11-29:
Perhaps this can be listed somewhere more prominently?

From the path:
ros2_control docs -> Core functionalities -> Concepts -> Hardware Components -> Migration from Foxy to Galactic

Are only the first and last element obvious.