ros-controls/ros2_controllers #274: Define a publish_rate for the diff_drive_controller

BenjaminHug8, 2021-11-30:
**Is your feature request related to a problem? Please describe.**
We want to limit the publication of the `/odom` and the `/tf` for the [diff_drive_controller](https://github.com/ros-controls/ros2_controllers/blob/foxy/diff_drive_controller/src/diff_drive_controller.cpp) to a certain rate. This change is for `foxy`, `galactic` and `master` branch.

**Describe the solution you'd like**
Define a `publish_rate` parameter and only publish a message if the passed time since the last publishing is greater than `1 / publish_rate`. I think it can simply be implemented as in the [diff_drive_controller for ROS](https://github.com/ros-controls/ros_controllers/blob/noetic-devel/diff_drive_controller/src/diff_drive_controller.cpp).

**Describe alternatives you've considered**
Any suggestion is welcomed.

**Additional context**
If the idea is accepted I will be happy to implement it!

destogl, 2021-12-09:
@BenjaminHug8 you can implement this :) and we will accept it

BenjaminHug8, 2021-12-13:
As we want this feature for `foxy`, `galactic` and `rolling`, should I target a particular branch (`foxy` or `master` branch) and the feature will be ported to the other branches or should I make a PR for each branch?

BenjaminHug8, 2022-01-13:
Done in #278

closed 2022-01-13 by BenjaminHug8