ros-controls/ros2_controllers #288: Ackerman Steering Controller

aboarya, 2022-01-26:
## Background

Aside from issue #264 have there been any attempts at an ackermann controller for ros2-controllers?  From what I can tell of that issue, the port is no longer being worked on.

Here's what I currently have:

1. A project with a URDF description that is gazebo valid and spawns a simple 4 wheel robot
2. Integration with gazebo_ros2_control linking to the ros2-control Diff Drive controller config
3. the small city world in Gazebo with a few cars, streets, stop signs, traffic lights, etc

My plan is to follow the instructions in this [comment](https://github.com/ros-controls/ros2_controllers/issues/264#issuecomment-996646394) and begin attempting to implement an ackerman steering controller.  

If I am successful in this, I may then open a PR for the controller, plus a PR in ros-control-demos with the content I described above.

Any input, feedback, or direction is highly welcomed!

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_controllers/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/foxy/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://ros-controls.github.io/control.ros.org/getting_started.html#compiling).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_controllers`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_controllers` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://ros-controls.github.io/control.ros.org/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/foxy/Tutorials.html)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

bmagyar, 2022-01-29:
This would be a very welcome change, thank you! Would you mind posting the same message in #264  that you are essentially planning to take over this effort to notify the people over there? And this issue could be closed as a duplicate of that.

aboarya, 2022-01-29:
Sure thing! Will do...

tonynajjar, 2022-04-21:
Hi @armuren, I'm sure many are impatiently waiting for your contributions! Could you share any updates on your progress?

tonynajjar, 2022-05-02:
We waited so impatiently that we decided to give it a try ourselves :smile: : https://github.com/ros-controls/ros2_controllers/pull/344/files
It's not exactly ackermann but I think with minor changes it could also work for ackermann as the kinematics are very similar as far as I know

armuren, 2022-05-16:
@tonynajjar my apologies! I got distracted and was not able to follow up on this.  I will close this PR

closed 2022-05-16 by armuren