ros-controls/ros2_controllers #290: joint_trajectory_controller should not use `now()` but rely on time passed in through the `update()` function

bmagyar, 2022-01-29:
## Background

joint_trajectory_controller should not use `now()` but rely on time passed in through the `update()` function. 

The goal of this ticket is to remove the usage of `node_->now()` from `joint_trajectory_controller.cpp` e.g. [here](https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L171) and replace it with the `time` parameter passed in [here](https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L110). 

Any usage of `node_->now()` that is in the same call tree as `update()`, meaning it is a function called within `update()` should also be eliminated. If needed, extend those function's parameters with `time`.

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_controllers/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/foxy/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://ros-controls.github.io/control.ros.org/getting_started.html#compiling).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_controllers`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_controllers` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://ros-controls.github.io/control.ros.org/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/foxy/Tutorials.html)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

Shubhanshu1902, 2022-01-29:
I would like to work on this issue @bmagyar

erickisos, 2022-02-11:
Hey Hi!

Sorry if this is a dumb question but I have one doubt, do we have a test suite to check if everything is working? I mean, to be sure that I can change the code without breaking anything?

Can you assign this to me as my first issue?

bmagyar, 2022-02-12:
We have quite a few tests for controllers, never enough of course but there should be plenty of coverage for this job

erickisos, 2022-02-12:
Amazing, thank you @bmagyar !

PR #296 (2022-03-12): Use time argument on update function instead of node time
Closes #290
closed 2022-03-12 by destogl