ros-controls/ros2_controllers #312: Ros2_control diff_drive cant turn in gazebo

RobertForristall, 2022-03-10:
Describe the bug
When working with the ros2_control package using the diff_drive_controller in gazebo the robot responds to the linear.x component of the twist message and drives forward; however it doesn't respond to the angular.z component of the message for turning.

To Reproduce
Steps to reproduce the behavior:

    Download the following package: https://drive.google.com/drive/folders/1SJmruuz5IH7Jz0EOMfMslK-AsFCxvWfa?usp=sharing

Important files in package:

    urdf of robot: resource/ezrassor.xacro
    gazebo file for robot: resource/ezrassor.gazebo
    ros2_controller config file: config/standard_rover/default_position_controllers.yaml
    launch file for simulation/robot: launch/simulation_launch.py

Build package:

    colcon build

Run the following launch command:

    ros2 launch ezrassor_arm_v2 simulation_launch.py rover_model:=standard control_methods:=autonomous

Wait for all nodes to launch

    (This will finish with a log to the console saying that the object detection and autonomous control nodes initalized.)

Publish the following message to console which will cause the robot to drive forward as intended:

    ros2 topic pub /ezrassor/diff_drive_controller/cmd_vel_unstamped geometry_msgs/Twist '{linear: {x: 1.0, y: 0.0, z: 0.0}, angular: {x: 0.0,y: 0.0,z: 0.0}}'

Publish the following message to console which will show the error as the robot will not turn as intended:

    ros2 topic pub /ezrassor/diff_drive_controller/cmd_vel_unstamped geometry_msgs/Twist '{linear: {x: 0.0, y: 0.0, z: 0.0}, angular: {x: 0.0,y: 0.0,z: 1.0}}'

Expected behavior
The robot should turn or at least show force being exerted to the wheels in an attempt to turn but no response is seen from the tires at all when given the angular.z component

Environment (please complete the following information):

    OS: Ubuntu 20.04 (Focal Fossa)
    Version Foxy

Additional context
Seems that there may be an issue with the communication of the angular.z component to the wheel drivers in gazebo. I have tried messing with the mu values as recommended in other issues like this but to no avail.

destogl, 2022-03-12:
> So what version is your all's stuff tested with, galactic? Just curious cause i may just upgrade the project to that version if its been tested and is stable

I was testing diff drive on rolling and galactic. There should be working.

And don't expect us to test your package. This sounds very much like wrong configuration. Check the output/publishers of DiffDrive controller and see what happens when you use "open_loop". Compare rviz and gazebo outputs.

destogl, 2022-06-20:
@RobertForristall any news on this issue?

JPDF, 2023-02-22:
Hi, I know this question has been opened for a while but I also have the same issue on foxy. I'm using open_loop and it appears that the odometry is updating accordingly but the wheels are not rotating when sending angular input. Only linear input makes the wheels turn. I can see that the odometry is updated in rviz and when manually inspecting odom but the velocity on the wheels are not considering the angular vector. I am using the gazebo_ros2_control package.

JPDF, 2023-02-22:
Well this is embarrassing... I misspelled **wheel_separation**/**wheel_seperation**... now it works great, thanks!

closed 2024-04-04 by christophfroehlich