ros-controls/ros2_controllers #337: Replace `RCLCPP_ERROR_STREAM` with `RCLCPP_ERROR` in gripper action controller

bmagyar, 2022-04-21:
Could we also change this macro to `RCLCPP_ERROR` instead of stream? There is no value in it, but it's confusing to read.

I know I am asking for something out of scope, but it seems there are only three places of this and it would be great to solve them right away.

_Originally posted by @destogl in https://github.com/ros-controls/ros2_controllers/pull/329#discussion_r852904860_

christophfroehlich, 2025-03-14:
https://github.com/search?q=repo%3Aros-controls%2Fros2_controllers%20RCLCPP_ERROR_STREAM&type=code

Vedant87, 2025-03-19:
Hello, I would like to contribute. To which branch should I open a PR ?

christophfroehlich, 2025-03-19:
please target the master branch, we will then backport this

closed 2025-03-20 by saikishor