ros-controls/ros2_controllers #347: No Output to Screen for JTC Commands in the Demos

jackcenter, 2022-05-02:
**Is your feature request related to a problem? Please describe.**
In the demos `ros2 launch ros2_control_demo_bringup test_joint_trajectory_controller.launch.py publisher_config:=rrbot_namespace_joint_trajectory_publisher.yaml`  doesn't output the commands to the screen like the other controllers do. 

**Describe the solution you'd like**
I was planning on adding the output to the test nodes [here](https://github.com/ros-controls/ros2_controllers/blob/master/ros2_controllers_test_nodes/ros2_controllers_test_nodes/publisher_joint_trajectory_controller.py)

**Describe alternatives you've considered**
Could assume the user knows how to echo the topic, or explain this in the tutorials. But I think printing the command to the console matches the other examples.

**Additional context**
Reference conversation can be found [here](https://github.com/ros-controls/ros2_control_demos/pull/170#pullrequestreview-956652878) in ros-controls/ros2_control_demos#170

jackcenter, 2022-05-02:
I am planning to work on this.

destogl, 2022-08-24:
Close because the node is moved to another repo and should be solved in #389

closed 2022-09-05 by destogl