ros-controls/ros2_controllers #353: ‘CallbackReturn’ in namespace ‘controller_interface’ does not name a type

tonynajjar, 2022-05-14:
**Describe the bug**
ros2_controllers won't build

**To Reproduce**
Steps to reproduce the behavior:
1. clone ros2_control and ros2_controllers on master branches
2. colcon build --packages-up-to ros2_controllers

several controllers fail to build because 

`error: ‘CallbackReturn’ in namespace ‘controller_interface’ does not name a type
`

**Expected behavior**
Successful build

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: Ubuntu 20
 - Version Rolling (docker container)

**Additional context**

bmagyar, 2022-05-15:
Worked for me yesterday.

Can you post exact steps to reproduce please?
Do you have ros2_control installed too? I found that to break builds sometimes

tonynajjar, 2022-05-15:
You're right, happens only when ros2_control is installed via apt as well.

The exact steps on a fresh rolling image are then:

1- apt install ros-rolling-ros2-control
2- clone ros2_control and ros2_controllers on master branches
3- colcon build (--packages-select diff_drive_controller --event-handlers console_direct+)

In the verbose mode, it shows the installed version "/opt/ros/rolling/include/controller_interface/controller_interface.hpp" being used. That's unexpected behavior though isn't it? How can I force it to use the built version?

AndyZe, 2022-05-23:
I think the conflict between a debian and a source build is a common, known issue with ROS2. I can't find a great reference but here's a similar issue. There, somebody says `ament` is the root cause.

https://github.com/ros-planning/moveit2/issues/796#issuecomment-967249746

With that said, I'm suffering from the same issue, even after removing the debians. Sigh. I'll report back if I find a solution.

destogl, 2022-07-27:
Closing because it should be an issue anymore.

closed 2022-07-27 by destogl