ros-controls/ros2_controllers #357: Odometry position approximation: exact integration / rungeKutta2 integration / rungeKutta4 integration?

engeen-nl, 2022-05-23:
Line 150 of odometry.cpp gives the comment:
`/// Exact integration (should solve problems when angular is zero):`
However, it is executed when `!(fabs(angular) < 1e-6)` as it is in the else clause. Is this supposed to be the other way around? Or is the comment wrong?
And maybe this discussion is better another place, but when I see the RungeKutta2 function, I'm not much of a Mathematician, but it looks a lot different from what I can find online about RungeKutta2. Is this a work in progress? And maybe we can implement RungeKutta4 instead for higher accuracy over longer distances? I understand with a high sample rate this shouldn't have much effect, but when driving fast / sampling little it will.
Background information: As I'm looking for ways to convert recorded motor controller positions to XY location, in order to visualise this, I try to see how ROS handles this.

closed 2025-05-30 by github-actions[bot]

Amronos, 2025-07-26:
Related to #271.
Please assign this to me. I will start working on it in about a week.

Amronos, 2025-08-20:
I am not much of a mathematician either, but I did try to learn a bit about Runge-Kutta. I decided to start the learning journey by using AI to learn some basics/pointers on which I could expand. During my conversation, I received the following response to one of my questions (I have deleted some repetitive parts of the response):

```
## 📌 Would RK4 Help Here?

RK4 shines when:

* The system dynamics are nonlinear, **and**
* You only know derivatives (like $dx/dt$, $dy/dt$, $d\theta/dt$) at discrete times,
* You need high accuracy because the **closed form solution is not available**.

But here:

* The differential drive kinematics are simple, and you already implemented the **exact analytic integration** for a timestep.
* RK4 would just approximate what you already computed exactly.
* In fact, using RK4 here would add unnecessary computation **without improving accuracy** (unless your assumption of constant velocity over `dt` is wrong).

---

## 📌 When Would RK4 Be Needed?

RK4 might be useful if:

* Your velocities are **not constant** during the timestep (e.g., high-frequency acceleration, nonlinear velocity profiles).
* You are integrating more complex models (like car-like robots with slip, or dynamics involving acceleration).
* You don’t have an exact closed-form solution for the motion model.

For example, if you were integrating:

$$
\dot{x} = v \cos(\theta), \quad \dot{y} = v \sin(\theta), \quad \dot{\theta} = \omega
$$

with **time-varying** $v(t), \omega(t)$, then RK4 is a good choice.

---
```

I am not at all sure of how accurate this response is, as it is from an AI. I will dig deeper into RK4 when I get the time to do so, but I don't think I am the best person for this job.
Could we continue with #1854, and somebody more experienced / I could get to this in a future PR?