ros-controls/ros2_controllers #372: JointTrajectoryController state interface

Cryoschrome, 2022-06-29:
### Environment
- ROS Distro: Humble
- OS Version: e.g. Ubuntu 22.04

### Description
I am using JointTrajectoryController, with velocity, acceleration command interface and position state interface, to execute the trajectory generated by MoveIt2!. The motor is connected to a teensy3.6 using [TeensyStep library](https://github.com/luni64/TeensyStep). I have create a actuator hardware interface. The controllers outputs the trajectory at a very high rate, which the motor cannot keep up with. For example if I create a trajectory to move the motor from 0.0 rads to 3.14 rads, the controllers very quickly outputs the trajectory and RViz reports that execution is complete while the motor position is still around 1 rads. 
Is there a way to slow down the rate at which the output is given or a ways to synchronize the command and state. 
Any response is appreciated
Thank you!

christophfroehlich, 2023-04-16:
Hi @SourabhPrasad 
sorry that nobody reacted to your question. Have you found the solution already (please share it) or is there still some open question?

SourabhPrasad, 2023-04-19:
Hi @christophfroehlich. I had paused development on this as I was not able to figure out the solution.
I will get back into it and let you know.

christophfroehlich, 2023-04-19:
Reading your description it seems that there is a misconfiguration. The controller is designed to sample/interpolate the trajectory at the current time step. If the trajectory is "too fast", there might be a problem with the time-axis from MoveIt. 
If the problem is the physical interface sending the comments via the hardware interface, then you can reduce the update rate of the controller manager via parameters.

christophfroehlich, 2023-04-19:
how do you connect the Teensy Library, via UART?

christophfroehlich, 2023-04-19:
how do you connect the Teensy Library, via UART?
Or maybe just the velocity limit of the stepper motor is lower than you've set it with moveit, so that the planned trajectory is just not feasible. The controller does not wait for it then

SourabhPrasad, 2023-04-22:
The connection between the PC and teensy was done through using MicroROS. Ideally, I would like to keep using micro-ros but I do believe that it does add additional latency to the system. I will try updating the update rate of the controller and velocity limit of the motor.

christophfroehlich, 2023-04-23:
Never used MicroROS, sorry. isn't its purpose to provide interfaces like topics etc to embedded hardware? how do you physically and logically connect ros2_control hw interface to MicroROS then?

SourabhPrasad, 2023-04-27:
So the physically connection was over USB. I had created additions publishers and subscribers within the hardware interface to set up communication with MicroROS.
I currently don't have the motor setup that I used during the development. So finding it a bit difficult to get it working. If possible could you point me towards how I simulate motor control with ros2_controller

christophfroehlich, 2023-04-27:
Having topics from the hardware interface is a possible way, e.g. also the Isaac Sim interface from MoveIt works in this way. Though I have no experience with that.
What do you mean by simulating your motors? Write a node attaching to the topics of your hardware interfaces? Or do you mean you want to simulate the robot with Gazebo or similar, and add actuator dynamics?

SourabhPrasad, 2023-04-28:
So we had a motor with a gearbox (80:1 ratio). I am not sure how setup the ratio with Gazebo

christophfroehlich, 2023-04-28:
- If you want to work with MoveIt and use Gazebo for robot simulation, you never need the gearbox ratio because you will stay in joint space everytime.  
- Adding actuator dynamics in Gazebo is not really straight forward, I wrote my own gazebo plugin inheriting gazebo_ros2_control plugin for this purpose. 
- When you switch the simulation to your real hardware, you can use the [transmission interface](https://github.com/ros-controls/ros2_control_demos/tree/master/example_8) within your custom hardware interface.

SourabhPrasad, 2023-05-01:
I will definitely look into the plugin. As the hardware interface was the part that I was struggling with. 
I had read about transmission interface previously, but wasn't able to figure out how to implement it. Was the demo you linked added recently?

Thank you @christophfroehlich for all this information. It is highly appreciated!

christophfroehlich, 2023-05-01:
yes, we added it just a couple of weeks ago.

christophfroehlich, 2024-02-15:
Closing due to inactivity

closed 2024-02-15 by christophfroehlich