ros-controls/ros2_controllers #381: What does imu_sensor_broadcaster controller do and how to use it?

robofoundry, 2022-07-09:
I have tried to read the code and follow the demo controllers repository but really cannot figure out what is the purpose of imu_sensor_broadcaster and how to use it. Can someone explain? 

I have used the diff_drive_controller several times and love the functionality and I was wondering what I can do with the out of the box imu_sensor_broadcaster. The only two parameters it takes is sensor name and frame_id unlike diff_drive_controller which takes so many and does a lot. 

Also, tried to read the code in the demo repository for rrbot with sensor to see if a custom imu broadcaster can be created and it looks like you are supposed to connect to real hardware sensor from within the derived class that implements hardware_interface::SensorInterface to read and publish the data. It makes sense for the custom controller but if the imu_sensor_broadcaster does something out of box would like to use that first. But there is hardly any documentation on that so any help in what it does and what config to feed it would be much appreciated. Thanks.

olivier-stasse, 2022-07-10:
One way to see the `imu_sensor_broadcaster` is a simple bridge between the ros2_control C++ interface world and the ros2 middleware world. Same for the `joint_state_brodcaster`.

From your question you may belooking for a state estimator that you could use for building a `controller` by composition of other ros2_controllers. Thanks to @destogl with the chained controller this is a functionality that is available.

robofoundry, 2022-07-10:
Thanks for the quick response @olivier-stasse. My hope was, if I wanted to integrate e.g. an MPU6050 or BMI160 IMU and I have c++ code to interface with the sensor, how would I use that with the imu_sensor_broadcaster? Or do I have to write a custom controller that implements SensorInterface in this scenario? A similar situation for the diff drive controller robot would be - just use the ros2_control out of box diff_drive_controller and then build the custom hardware interface e.g. for dynamixel
and then in the urdf configure the robot to use that dynamixel hardware interface and then it all works together.

Basically, high level steps to get it to work with diff drive controller and custom hardware interface would be something like this:
1. create a custom hardware interface e.g. dynamixel or odrive etc
2. setup controller.yaml file so it has diff_drive_controller and all its params
3. in URDF configure the ros2_control node which refers to the hardware interface and all the parameters it needs
4. in launch file start all the appropriate controllers 

would the steps to use the imu_sensor_broadcaster similar and still requires that you write that custom hardware interface with the imu?

I'm trying to figure out how to use it as a bridge if I have a hardware library to read the sensor and may be that's not what it does. And I still need to write code for a custom controller that reads the sensor and publishes the sensor data. The question in that case would be what is the value that imu_sensor_broadcaster brings where the custom controller could directly publish the sensor data.

Hope I'm able to articulate the question better.

olivier-stasse, 2022-07-10:
> Thanks for the quick response @olivier-stasse. 

You are welcome.

> My hope was, if I wanted to integrate e.g. an MPU6050 or BMI160 IMU and I have c++ code to interface with the sensor, how would I use that with the imu_sensor_broadcaster? Or do I have to write a [...]

You will need to implement a SensorInterface for your IMU.
> robot would be - just use the ros2_control out of box diff_drive_controller and then build the custom hardware interface e.g. for dynamixel and then in the urdf configure the robot to use that dynamixel hardware interface and then it all works together.

There is already a ros2_control driver for dynamixel:
https://github.com/youtalk/dynamixel_control
> Basically, high level steps to get it to work with diff drive controller and custom hardware interface would be something like this:
>
>     1. create a custom hardware interface e.g. dynamixel or [...]

The imu_sensor_broadcaster will read the sensor value that you provided through your SensorInterface and publish it in a topic. It can be then used by the Navigation stack to update the odometry. If you have a laser or a RGB-D camera then this is allowing to use the acml algorithm. This is what you gain by going through ros2_control 
> I'm trying to figure out how to use it as a bridge if I have a hardware library to read the sensor and may be that's not what it does. And I still need to write code for a custom controller that [...]

destogl, 2022-09-26:
I think the question is answered, therefew closing this.

closed 2022-09-26 by destogl