ros-controls/ros2_controllers #394: Discourage changing the number of DoFs / joints for controller after those are initialized?

destogl, 2022-07-27:
Discussing parameters handling with @tylerjw and @pac48 it seems that resetting the number of DoF in `on_configure` in some controllers only complicates the management of those. This does not make much sense because if one needs to change the number of DoF in a controller, another instance of the controller should be used. 

Recommended resolution for all (standard) controller in this package:
1. Read the number of `joints`/`joint_names` parameters in the `on_init` method.
2. Remove updates of parameters from `on_configure` method.

AndyZe, 2022-07-27:
This will also improve realtime performance. If you look at the JTC, all of the dynamic memory allocation occurs because it is possible for `dof` to change size. (Noted in #393)

bmagyar, 2022-07-27:
There should be a whole bunch of parameters I'd discourage people from changing, including DOF