ros-controls/ros2_controllers #404: Crash in force_torque_sensor_broadcaster

fmauch, 2022-08-02:
**Describe the bug**
Since we finally switched towards this repo's force_torque_sensor_broadcaster inside the ur_robot_driver, I've noticed crashes inside the controller on ROS galactic.

**To Reproduce**
Steps to reproduce the behavior:
1. Install ur_robot_driver (sudo apt install ros-galactic-ur-robot-driver)
2. Start the robot (simulation does it as well: `docker run --rm -it -p 5900:5900 -p 6080:6080 --name ursim universalrobots/ursim_e-series`)
3. Launch the driver: `ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=172.17.0.2 launch_rviz:=false`
4. The driver will die once the fts_broadcaster is loaded:

```
[ur_ros2_control_node-1] [INFO] [1659467343.755580519] [controller_manager]: Loading controller 'force_torque_sensor_broadcaster'
[spawner-8] [INFO] [1659467343.762149823] [spawner_force_torque_sensor_broadcaster]: Loaded force_torque_sensor_broadcaster
[ERROR] [ur_ros2_control_node-1]: process has died [pid 24783, exit code -11, cmd '...'].
```
 
When not spawning the controller directly, I can successfully load the fts_broadcaster, but as soon as I interact with it (e.g. `ros2 control list_controllers`) the same crash happens.

**Expected behavior**
Well, it would be nice if the controller would not crash and correctly report the wrench.

**Screenshots**
I built ros2_control, ros2_controllers and the driver from source with debug information and this was the traceback:

```
[INFO] [1659467582.672577931] [controller_manager]: Loading controller 'force_torque_sensor_broadcaster'

Thread 1 "ur_ros2_control" received signal SIGSEGV, Segmentation fault.
0x00007fffe4768fca in force_torque_sensor_broadcaster::ForceTorqueSensorBroadcaster::state_interface_configuration (this=0x7fffbc004750) at /usr/include/c++/9/bits/stl_vector.h:94
94              _Vector_impl_data() _GLIBCXX_NOEXCEPT
(gdb) bt
#0  0x00007fffe4768fca in force_torque_sensor_broadcaster::ForceTorqueSensorBroadcaster::state_interface_configuration (this=0x7fffbc004750) at /usr/include/c++/9/bits/stl_vector.h:94
#1  0x00007ffff7f4b1c7 in controller_manager::ControllerManager::list_controllers_srv_cb (this=0x55555558f560, 
    response=std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> >> (use count 3, weak count 0) = {...}) at /usr/include/c++/9/bits/shared_ptr_base.h:1020
#2  0x00007ffff7f64d9d in std::__invoke_impl<void, void (controller_manager::ControllerManager::*&)(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >), controller_manager::ControllerManager*&, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > > > (__t=<optimized out>, __f=<optimized out>) at /usr/include/c++/9/bits/shared_ptr_base.h:756
#3  std::__invoke<void (controller_manager::ControllerManager::*&)(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >), controller_manager::ControllerManager*&, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > > > (__fn=<optimized out>) at /usr/include/c++/9/bits/invoke.h:95
#4  std::_Bind<void (controller_manager::ControllerManager::*(controller_manager::ControllerManager*, std::_Placeholder<1>, std::_Placeholder<2>))(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >)>::__call<void, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >&&, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >&&, 0ul, 1ul, 2ul>(std::tuple<std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >&&, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >&&>&&, std::_Index_tuple<0ul, 1ul, 2ul>) (__args=..., this=<optimized out>) at /usr/include/c++/9/functional:400
#5  std::_Bind<void (controller_manager::ControllerManager::*(controller_manager::ControllerManager*, std::_Placeholder<1>, std::_Placeholder<2>))(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >)>::operator()<std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >, void>(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >&&, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >&&) (this=<optimized out>) at /usr/include/c++/9/functional:484
#6  std::_Function_handler<void (std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >), std::_Bind<void (controller_manager::ControllerManager::*(controller_manager::ControllerManager*, std::_Placeholder<1>, std::_Placeholder<2>))(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >&&, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >&&) (__functor=..., __args#0=..., __args#1=...) at /usr/include/c++/9/bits/std_function.h:300
#7  0x00007ffff7f693fb in std::function<void (std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >)>::operator()(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >) const (
    __args#1=std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> >> (empty) = {...}, 
    __args#0=std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> >> (empty) = {...}, this=0x55555561adc0) at /usr/include/c++/9/bits/std_function.h:683
#8  rclcpp::AnyServiceCallback<controller_manager_msgs::srv::ListControllers>::dispatch (response=..., request=..., request_header=..., this=0x55555561adc0) at /opt/ros/galactic/include/rclcpp/any_service_callback.hpp:94
#9  rclcpp::Service<controller_manager_msgs::srv::ListControllers>::handle_request (this=0x55555561ad90, request_header=std::shared_ptr<rmw_request_id_t> (use count 3, weak count 0) = {...}, request=...)
    at /opt/ros/galactic/include/rclcpp/service.hpp:339
#10 0x00007ffff7db966d in rclcpp::Executor::<lambda()>::operator() (__closure=<optimized out>) at /usr/include/c++/9/ext/atomicity.h:96
#11 std::_Function_handler<void(), rclcpp::Executor::execute_service(rclcpp::ServiceBase::SharedPtr)::<lambda()> >::_M_invoke(const std::_Any_data &) (__functor=...) at /usr/include/c++/9/bits/std_function.h:300
#12 0x00007ffff7dba014 in std::function<void ()>::operator()() const (this=0x7fffffffaa80) at /usr/include/c++/9/bits/std_function.h:683
#13 take_and_do_error_handling(const char *, const char *, std::function<bool()>, std::function<void()>) (action_description=0x7ffff7e85760 "taking a service server request from service", 
    topic_or_service_name=0x555555625690 "/controller_manager/list_controllers", take_action=..., handle_action=...) at ./src/rclcpp/executor.cpp:563
#14 0x00007ffff7dba28c in rclcpp::Executor::execute_service (service=std::shared_ptr<rclcpp::ServiceBase> (use count 3, weak count 1) = {...}) at /usr/include/c++/9/bits/shared_ptr_base.h:1020
[... cut ...]
I am not sure if this is **really** related to this repo, but it seems like this is something to be handled outside of our driver. Also it would be noteworthy, that the problem seems not to exist on ubuntu 22.04 with ROS2 humble or rolling (testing repo versions).

I am also not sure whether this worked at any time. We've fixed loading the correct controller quite [recently](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/commit/a850e6b8d1d91b40b580dd9729a0d24d0e8883b7) and I am not sure whether I've only tested whether the error about loading the driver went away.

fmauch, 2022-08-02:
Here's a traceback with full debug flags:

```
#0  force_torque_sensor_broadcaster::ForceTorqueSensorBroadcaster::state_interface_configuration (this=0x7fffa40057a0)
    at /home/mauch/robot_folders/checkout/ur_galactic_semi_binary/colcon_ws/src/ros2_controllers/force_torque_sensor_broadcaster/src/force_torque_sensor_broadcaster.cpp:141
#1  0x00007ffff7ebe953 in controller_manager::ControllerManager::list_controllers_srv_cb (this=0x555555596560, 
    response=std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> >> (use count 3, weak count 0) = {...})
    at /home/mauch/robot_folders/checkout/ur_galactic_semi_binary/colcon_ws/src/ros2_control/controller_manager/src/controller_manager.cpp:877
#2  0x00007ffff7f07c25 in std::__invoke_impl<void, void (controller_manager::ControllerManager::*&)(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >), controller_manager::ControllerManager*&, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > > > (__f=
    @0x555555662440: (void (controller_manager::ControllerManager::*)(controller_manager::ControllerManager * const, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >)) 0x7ffff7ebe350 <controller_manager::ControllerManager::list_controllers_srv_cb(std::sha--Type <RET> for more, q to quit, c to continue without paging--
red_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >)>, __t=@0x555555662450: 0x555555596560)
    at /usr/include/c++/9/bits/invoke.h:73
#3  0x00007ffff7efe81d in std::__invoke<void (controller_manager::ControllerManager::*&)(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >), controller_manager::ControllerManager*&, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > > > (__fn=
    @0x555555662440: (void (controller_manager::ControllerManager::*)(controller_manager::ControllerManager * const, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >)) 0x7ffff7ebe350 <controller_manager::ControllerManager::list_controllers_srv_cb(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >)>) at /usr/include/c++/9/bits/invoke.h:95
#4  0x00007ffff7ef768d in std::_Bind<void (controller_manager::ControllerManager::*(controller_manager::ControllerManager*, std::_Placeholder<1>, std::_Placeholder<2>))(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListCont--Type <RET> for more, q to quit, c to continue without paging--
rollers_Response_<std::allocator<void> > >)>::__call<void, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >&&, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >&&, 0ul, 1ul, 2ul>(std::tuple<std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >&&, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >&&>&&, std::_Index_tuple<0ul, 1ul, 2ul>) (this=0x555555662440, __args=...)
    at /usr/include/c++/9/functional:400
#5  0x00007ffff7ef2373 in std::_Bind<void (controller_manager::ControllerManager::*(controller_manager::ControllerManager*, std::_Placeholder<1>, std::_Placeholder<2>))(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >)>::operator()<std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >, void>(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >&&, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >&&) (this=0x555555662440)
    at /usr/include/c++/9/functional:484
#6  0x00007ffff7eeca4f in std::_Function_handler<void (std::shared_ptr<controller_manager_msgs::srv::ListControllers_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >), std::_Bind<void (controller_manager::ControllerManager::*(controller_--Type <RET> for more, q to quit, c to continue without paging--
[... cut ...]
    this=this@entry=0x5555555921c0, any_exec=...)
    at /usr/include/c++/9/ext/atomicity.h:96
#16 0x00007ffff7b694bb in rclcpp::executors::MultiThreadedExecutor::run (
    this=0x5555555921c0)
    at ./src/rclcpp/executors/multi_threaded_executor.cpp:106
#17 0x00007ffff78d7de4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#18 0x00007ffff79eb609 in start_thread (arg=<optimized out>)
    at pthread_create.c:477
#19 0x00007ffff7713133 in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

fmauch, 2022-08-02:
Following the traceback this seems to be related to https://github.com/ros-controls/ros2_control/pull/580. I could cherry-pick that [commit](https://github.com/ros-controls/ros2_control/commit/223b2593b09e3c2b453d8c02f7f402d590316124) into galactic directly and it seemed to fix the issue for me. So, I guess it is indeed the wrong repo and this issue should be inside `ros2_control`.

@maintainers: Is there any chance that this will get fixed before the next galactic sync (which freezes tomorrow)? Otherwise we would [disable](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/461) the fts_broadcaster in the next release.

bmagyar, 2022-08-02:
I think anything for tomorrow is a fairly short notice but we can certainly try. I can bring attention to this issue to argue it being included while things are being frozen.

bmagyar, 2022-08-03:
https://github.com/ros/rosdistro/pull/34052

destogl, 2022-08-04:
This is some great debugging! Thanks!

fmauch, 2022-08-05:
@bmagyar Thank you for taking such quick action. I haven't been actively working on this the last two days, but I followed along the successful merge into the rosdistro.

I've now confirmed that this resolves the issue described above using the binary packages from the testing repo, so the version synced today will work for users :-)

Closing this as resolved.

closed 2022-08-05 by fmauch