ros-controls/ros2_controllers #406: Joint State Broadcaster Renames Joints

swiz23, 2022-08-04:
**Describe the bug**
The joint_state_broadcaster seems to rename joints (by tacking on '/position' or '/velocity' to the name). For example, this is what my ros2 controllers yaml looks like

```
$(var controller_ns):
  controller_manager:
    ros__parameters:
      update_rate: 30 # Hz

      position_traj_controller:
        type: joint_trajectory_controller/JointTrajectoryController

      streaming_position_controller:
        type: position_controllers/JointGroupPositionController

      joint_state_broadcaster:
        type: joint_state_broadcaster/JointStateBroadcaster

  position_traj_controller:
    ros__parameters:
      command_interfaces:
        - position
      state_interfaces:
        - position
        - velocity
      joints:
        - joint_1
        - joint_2
        - joint_3
        - joint_4
        - joint_5
        - joint_6
      allow_partial_joints_goal: true
```
So when the joint_state_broadcaster publishes to the <controller_ns>/joint_state topic, I'd expect the names to be equivalent to the 'joint_1, joint_2' etc... However, this is not the case. Instead the message that is published looks like the following:

```
---
header:
  stamp:
    sec: 1659630741
    nanosec: 154901218
  frame_id: ''
name:
- joint_6/velocity
- joint_6/position
- joint_1/position
- joint_5/velocity
- joint_5/position
- joint_4/velocity
- joint_2/velocity
- joint_4/position
- joint_3/velocity
- joint_1/velocity
- joint_3/position
- joint_2/position
position:
- .nan
- -1.5708
- -1.5708
- .nan
- 1.5708
- .nan
- .nan
- 0.0
- .nan
- .nan
- -1.41372
- -1.72788
velocity:
- 0.0
- .nan
- .nan
- 0.0
- .nan
- 0.0
- 0.0
- .nan
- 0.0
- 0.0
- .nan
- .nan
effort:
- .nan
- .nan
- .nan
- .nan
- .nan
- .nan
- .nan
- .nan
- .nan
- .nan
- .nan
- .nan
---
```
It's almost as if the state interface name is being published instead of the joint name...

**To Reproduce**
Steps to reproduce the behavior:
N/A

**Expected behavior**
Instead of the name being 'joint_6/velocity', it should be 'joint_6', etc...

**Screenshots**
N/A

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version: Humble
 - joint_state_broadcaster Version: 2.8.0-1jammy.20220714.173851 
- Anything that may be unusual about your environment: This is within a docker container

**Additional context**
N/A

swiz23, 2022-08-04:
Closing since it turned out the Docker container I was using built ros2_controllers from source using tag 2.6. I just tried building ros2_controllers from source for version 2.8 and everything seems to work as expected.

closed 2022-08-04 by swiz23

bmagyar, 2022-08-04:
Glad to hear, thanks!