ros-controls/ros2_controllers #411: Units of velocity feedback in diff_drive_controller

macstepien, 2022-08-10:
Currently velocity feedback in diff_drive_controller is expected to be in m/s. Taking into account that position feedback is in radians and velocity commands are in rad/s it seems quite odd. I think that it will be better to also use rad/s there, following change should be enough:
```
odometry_.updateFromVelocity(
        left_feedback_mean * left_wheel_radius * period.seconds(), right_feedback_mean * right_wheel_radius * period.seconds(), time);
```

destogl, 2022-09-26:
This sounds very plausible. You can open a PR and execute tests and see what they are saying. Maybe there are frew other places where this feedback is used and should be also adjusted.

macstepien, 2022-10-17:
I executed tests locally, but noticed that even on the current `master` branch `diff_drive_controller` tests sometimes failed. It seems that `correct_initialization_using_parameters` is not really deterministic or there was something wrong with my setup (it would be nice to run CI on the PR that I created). Additionally I looked into this test and it doesn't look like my change will be executed there. I also checked if any other change will be necessary, but it seems that it was the only place.

PR #452 (2022-12-03): Change units of velocity feedback in diff_drive_controller #behaviour-change
Closes #411 
Adds multiplying velocity feedback by wheel radius. When the `position_feedback` option is used, feedback is multiplied by radius (it is expected to be in `rad`). This PR changes velocity feedback to be also multiplied by radius, so that units used will be more consistent.
closed 2022-12-03 by destogl